arduino/Arduino #8663: Serial plotter standalone?

rin67630, 2019-03-16:
The serial plotter from the Arduino IDE is a really nice thing.
I am using it right often on weak Raspberry Pi devices.

The only caveat is that to use it, I must jettison the bulky IDE, which takes unnecessarily a lot of resources and time to start.

Have we got a way to start the serial plotter standalone?
If not, would it be a big thing to get this done? (I mean not a complete autonomous program, but just the bare minimum to start just the serial plotter directly within the existing IDE installed folders.)

Thank you for considering.

rin67630, 2019-08-14:
Any update on that issue?

madsdyd, 2020-04-07:
Skimming the source code, I would think that starting the serial plotter without running the IDE is non-trivial to add to the existing code base. E.g. the code for setting the port is part of the Arduino IDE. Given enough requirements, you end up effectively starting the IDE anyway (even if it is hidden).