arduino/Arduino #8664: Tools Menu Gets Stuck On Linux with Focus Follows Mouse

PaulStoffregen, 2019-03-16:
On Linux with a desktop configured for focus follows mouse, the Tools>Boards and Tools>Ports menus seem to get stuck & become unresponsive if the cursor moves outside the menu and another window get focus.

Here's a video demo
https://www.youtube.com/watch?v=H_Ue1jXwlF8

facchinm, 2019-03-18:
Wow, I never hear of focus following the mouse. Is it a regression or did it always behave like this?

PaulStoffregen, 2019-03-18:
Looks like the problem goes all the way back to Arduino 1.0.6.  I was unable to run 1.0.5.

Maybe only very old Unix folks know about & use focus follows mouse these days?  Likewise for the middle mouse button for select-paste (instead of copy-paste).  Ok to consider this one a pretty obscure bug...