arduino/Arduino #8665: Tired of resizing window

qmr, 2019-03-16:
Tired of resizing window every single time I open the application.

Why don't you remember window size and use last size?

Or at least make the default usable.

per1234, 2019-03-16:
Which version of the Arduino IDE are you using?

qmr, 2019-03-16:
188 on Windows N 7 64 at the moment

per1234, 2019-03-16:
Closing as duplicate of https://github.com/arduino/Arduino/issues/8337

Please update to Arduino IDE 1.8.9, which has the bug fixed.

In the future, please do a quick test of the hourly build of the Arduino IDE before submitting a bug report to be sure that it hasn't already been fixed:
https://www.arduino.cc/en/Main/Software#hourly

closed 2019-03-16 by per1234

MSBGit, 2020-11-11:
I am running 1.8.13 (Windows Store 1.8.42.0) on Windows 10 and I am seeing the problem where the IDE doesn't save it's size or position and on startup uses a very small window. I exit the IDE using File -> Quit. I have a 3-headed system, 2 heads are 4K monitors and one is 1080p. The primary monitor is 4K and that's where the IDE always starts up, no matter which head I exit the IDE form (so I always have to drag to another desktop and resize the IDE when I start it).

Any debugging that I can do to help you figure this out?

[image]

[image]

facchinm, 2020-11-12:
@MSBGit I think the problem might be in these lines
https://github.com/arduino/Arduino/blob/a056a5fdc751b933453830434e36aeffae7512d5/app/src/processing/app/Base.java#L618-L623
Java is known to handle terribly multihead systems, so we avoid restoring the windows position if ANY of the dimensions has changed. 
This is a bit of a brutal workaround to avoid working with 2 monitors, moving the IDE (or the serial monitor) on the second one, close the IDE, detach the monitor, reopen the IDE, no serial monitor and no way to recover it since it's being painted offscreen.
Any patch to make it better would be greatly appreciated :slightly_smiling_face: