arduino/Arduino #8671: How to make sure arduino_debug.exe to use usbasp to upload my code.

m0923678421, 2019-03-18:
No problem with this commandline when I adjust programmer item to the usbasp in Arduino IDE first.

"C:\Program Files (x86)\Arduino\arduino_debug.exe"  --upload  --board arduino:avr:uno --verbose "$(FULL_CURRENT_PATH)" --useprogrammer

But I don't want to open Arduino IDE and also can change other programmer.
Now I try to add it in notepad++.
Anyone know how to do it?
Please help!!

per1234, 2019-03-19:
Use `--pref` to set the `programmer` property:
```
"C:\Program Files (x86)\Arduino\arduino_debug.exe" --pref programmer=arduino:usbasp --upload --board arduino:avr:uno --verbose "$(FULL_CURRENT_PATH)" --useprogrammer
```
You can find the available preferences by looking in preferences.txt, which is at the the location shown at the line following **File > Preferences > More preferences can be edited directly in the file** in the Arduino IDE. The `programmer` property is defined like this:
```
programmer={vendor}:{programmer ID}
```
`{vendor}` is defined by the name of the [vendor folder](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#hardware-folders-structure) of the hardware core.

You will find the programmer IDs listed in the programmers.txt of each hardware core ([example](https://github.com/arduino/ArduinoCore-avr/blob/master/programmers.txt)). 

---
In the future, please submit this sort of question to the Arduino Forum:
http://forum.arduino.cc/
This issue tracker is to be used only for reporting bugs and making feature requests, not providing support.

closed 2019-03-19 by per1234

m0923678421, 2019-03-19:
> Use `--pref` to set the `programmer` property:
>
> ```
> "C:\Program Files (x86)\Arduino\arduino_debug.exe" --pref programmer=arduino:usbasp --upload --board arduino:avr:uno --verbose [...]

I'm sorry for ask at wrong way and thank for you time.
It's work well now.

Thank you.