arduino/Arduino #8673: [Library Manager] Please include Sparkfun Qwiic Keypad Arduino Library

lewispg228, 2019-03-18:
Please add this SparkFun Qwiic Keypad Arduino Library to the Library Manager.
[https://github.com/sparkfun/SparkFun_Qwiic_Keypad_Arduino_Library](https://github.com/sparkfun/SparkFun_Qwiic_Keypad_Arduino_Library)

Thanks

-Pete

cmaglie, 2019-03-18:
mmm are you sure the link is correct?

lewispg228, 2019-03-18:
@cmaglie Sorry about that... I forgot to make it public. Should be gtg now. Thanks for your qwiic response!

cmaglie, 2019-03-29:
done!

closed 2019-03-29 by cmaglie

lewispg228, 2019-03-29:
Awesome. Thanks so much @cmaglie . We are very excited to see this pop up in the Arduino Library manager. Wahoo!!

lewispg228, 2019-04-23:
Hi There,
Library Manager doesn't seem to be recognizing our releases on this library. Is there something we're doing incorrectly?

Note, our latest release was created 12 days ago, so we're assuming the manager has already attempted to update.

https://github.com/sparkfun/SparkFun_Qwiic_Keypad_Arduino_Library/releases

Let us know and thanks!
-Pete