arduino/Arduino #8678: It's possible to join a search box to serial monitor

offcircuit, 2019-03-19:
Hi again,

Just this question/idea.

"It's possible to join a search box to serial monitor"

Thanks,
Helder Rodrigues

Pieter12345, 2019-03-25:
On first thought, I like this idea. However, I cannot think of cases where I would use this. What I often do is copy the output to notepad++ to not only search, but also temporarily store it (to process later or compare with another execution). If the serial monitor would come with a feature where the current output could be copied into some new tab, then a search box would make more sense to me. What is your use case and how do you think about this additional idea?

Side note: It might me tough to come up with desired behavior of a search box that searches in a text pane that is continuously being updated with auto-scroll active.

offcircuit, 2019-03-25:
Thanks a lot for the atencion,

Helder Rodrigues

closed 2019-03-25 by offcircuit

per1234, 2022-12-01:
Request now tracked at https://github.com/arduino/arduino-ide/issues/1553