arduino/Arduino #8680: Option to clean project (like have in Eclipse and Plataformio

JoaoLopesF, 2019-03-19:
Hi, I am be a Arduino Libraries too,

And I have some options (#defines) to the programmer,
can change, e.g. disable all debug, to affect behaviors on library.

But it not affect the compiled binary, sometimes, 
due Arduino IDE used cached compiled files of libraries.
E.g.:
`
Compiling library "RemoteDebug"
Using previously compiled file: /var/folders/6g/c30wh2nx7t15v_6cl5txwdyw0000gp/T/arduino_build_88941/libraries/RemoteDebug/RemoteDebugWS.cpp.o
Using previously compiled file: /var/folders/6g/c30wh2nx7t15v_6cl5txwdyw0000gp/T/arduino_build_88941/libraries/RemoteDebug/RemoteDebug.cpp.o
Using previously compiled file: /var/folders/6g/c30wh2nx7t15v_6cl5txwdyw0000gp/T/arduino_build_88941/libraries/RemoteDebug/utility/WebSocketsClient.cpp.o
Using previously compiled file: /var/folders/6g/c30wh2nx7t15v_6cl5txwdyw0000gp/T/arduino_build_88941/libraries/RemoteDebug/utility/WebSockets.cpp.o
Using previously compiled file: /var/folders/6g/c30wh2nx7t15v_6cl5txwdyw0000gp/T/arduino_build_88941/libraries/RemoteDebug/utility/WebSocketsServer.cpp.o
Using previously compiled file: /var/folders/6g/c30wh2nx7t15v_6cl5txwdyw0000gp/T/arduino_build_88941/libraries/RemoteDebug/utility/libb64/cencode.c.o
Using previously compiled file: /var/folders/6g/c30wh2nx7t15v_6cl5txwdyw0000gp/T/arduino_build_88941/libraries/RemoteDebug/utility/libb64/cdecode.c.o
Using previously compiled file: /var/folders/6g/c30wh2nx7t15v_6cl5txwdyw0000gp/T/arduino_build_88941/libraries/RemoteDebug/utility/libsha1/libsha1.c.o
`
In arduino one way I knowed to do it is changing board options in Tool menu.
Or hard way, rm -rf in build directory and reload the Arduino

Please add this clean in menu "Tools"

Regards

facchinm, 2019-03-20:
Hi @JoaoLopesF ,
we take care of dependencies in `arduino-builder` and just rebuild the files that are affected by other files' changes. To do this, all .d files produced by gcc are parsed and decide if the chache can be used based on the timestamp of the last modification.
If you have a working example of this mechanism failing, please let us know :slightly_smiling_face:

JoaoLopesF, 2019-03-20:
Hi @facchinm,
I knowed that is made to compile more fast.
But all modern IDE, as Eclipse and VSCode as a "clean project option"

This is important to force a new compilation.

Example:

in ino file:

`#define MEMORY_SAVER true // low features to save memory`

in cpp library file:
`
#ifndef MEMORY_SAVER
// do a lot of things
#endif
`
In C/C++ only way to determine that will be compiled or not,
is with a preprocessor diretives (#if,#ifdef,#define, etc).
I know that the Arduino not recomend use this,
but to advanced libraries it is very important

Regards

per1234, 2019-03-21:
@JoaoLopesF the .ino file and the .cpp file are separate translation units so a `#define` in the .ino file will not have any effect on the .cpp file. The exception would be if you were to add an `#include` directive for the .cpp file in the .ino file after the `#define`, but I've never seen a case where it made sense to `#include` a .cpp file.

Please provide a Minimal, Complete, and Verifiable Example that demonstrates the issue you have found, along with a complete set of steps we can use to reproduce the issue.

>But all modern IDE, as Eclipse and VSCode as a "clean project option"

The difference between those and the Arduino IDE is that the Arduino IDE's main focus is to be beginner friendly. For this reason, we have to be careful to keep the UI as simple as possible, while still trying to provide a IDE that is capable of doing advanced things. It would be better to fix any issues that would make a clean build necessary instead of adding another UI element as a "band-aid".

The only outstanding issue that has made me resort to forcing a full recompile via the board change workaround you described is that compiler warnings are not displayed for files which are not compiled due to having been cached in the prior compilation. However, there is already a proposed fix for that: https://github.com/arduino/arduino-builder/pull/302.

per1234, 2019-09-16:
Closing due to lack of response. @JoaoLopesF if you ever feel like providing the requested information, reply here with it and I'll be happy to re-open the issue.

closed 2019-09-16 by per1234