arduino/Arduino #8683: [Library Manager] Please add ESPNextion

JoaoLopesF, 2019-03-20:
Hi,

Please add my new library to the library manager:

https://github.com/JoaoLopesF/ESPNextion

Thanks and congrats to excelent tool.

per1234, 2019-03-21:
In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Make sure that the `version` value matches the tag of the release that contains it.

After adding the library.properties file, you will need to either create a new tag/release, or else redo the existing tag/release.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

JoaoLopesF, 2019-03-21:
Hi, I forgotten this. Thanks

JoaoLopesF, 2019-03-23:
Hi, please verify again and add this lib:

https://github.com/JoaoLopesF/ESPNextion

Thanks a lot

per1234, 2019-03-23:
Each library in the Library Manager index must have a unique `name` value in library.properties. There is already a library in the Library Manager index with the same `name` value as in your library.properties: `RemoteDebug`.
https://github.com/JoaoLopesF/ESPNextion/blob/v0.9.0/library.properties#L1

Please change the `name` value in the ESPNextion library.properties to a unique value, which is compliant with the library specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
Then create a new tag/release (after updating the library.properties `version` value) or redo the existing tag/release.

You can check which names are already taken in the Arduino IDE's Library Manager (Sketch > Include Library > Manage Libraries) or here:
http://downloads.arduino.cc/libraries/library_index.json

After you have done this, please comment here so we'll know we can proceed.

per1234, 2019-07-26:
Since there has been no action to make the necessary changes after 4 months, I'm going to go ahead and close this. @JoaoLopesF if you ever get your library compliant with the Library Manager requirements please comment here and I'll re-open the request.

closed 2019-07-26 by per1234