arduino/Arduino #8687: "Sketch -> Include Library" menu scales while scrolling

Pieter12345, 2019-03-21:
When using the mouse scroll wheel to scroll from the top of the "Sketch -> Include Library" to the bottom, there is one point where the scaling changes.
When in the left situation and scrolling 1 'tick' down, the right situation occurs (where all items take slightly more vertical space):
[image]

facchinm, 2019-03-27:
I always wanted to solve this issue but ended up thinking that it must be either a Swing bug or either a side effect of something we are doing when creating the menu. 
Anyway, I think it's very low priority since it doesn't affect the functionality, just a bit annoying :slightly_smiling_face:

Pieter12345, 2019-03-27:
The issue seems to be that the view shows a fixed amount of components in the fixed menu height and that the gray line components have a different height than text components. When a scroll removes a text component and inserts a gray line, this makes the total hight of elements in the list smaller, but it's still being spread out over the fixed height of the menu, causing this weird scaling to happen.