arduino/Arduino #8694: [Library Manager] add DualVNH5019MotorShieldMod3

photodude, 2019-03-25:
Please consider adding https://github.com/photodude/DualVNH5019MotorShieldMod3 to the library manager index

This is a custom library for the Arduino that interfaces with two Pololu Dual VNH5019 Motor Driver Shields with an Arduino Mega. It makes it simple to drive four brushed, DC motors from 2 shields on an Arduino Mega. 

This is a modification of the pololu library pololu/dual-vnh5019-motor-shield extending the functions to work with two shields and 4 motors on an arduino mega. The library is referenced in on the Pololu resource page https://www.pololu.com/product/2507/resources

cmaglie, 2019-03-29:
Done!

closed 2019-03-29 by cmaglie

photodude, 2019-03-29:
@cmaglie thank you.