arduino/Arduino #8695: can't use Mac-compressed zip themes

R0dri, 2019-03-25:
I download and applied succesfully ArduinoDark theme from a .zip downloaded somewhere online. I tried to zip other themes from https://github.com/AbhishekGhosh/Arduino-Themes yet unsuccessful. After a lot of trial and error, found out that uncompressing the functioning .zip file and compressing it again renders it invisible for the IDE. Even under the same name. Changing the name while to the .zip downloaded file has no effect, IDE recognizes it after reopening preferences.

Another thing I have noticed is that the un-compressed and re-compressed file is quite larger being 70kb the downloaded one and 86kb the re compressed one. Maybe something about apple zipping protocol?

per1234, 2019-03-25:
A tricky thing about the Arduino IDE's .zip theme system is that it requires the theme to be in the root of the .zip folder, rather than in a subfolder (which is how the overwhelming majority of .zip files are structured). Are you sure that the folder structure of the working and non-working .zip files are exactly the same?

Another thought: I know that .zip files created on macOS will have a `__MACOSX` folder added. This, and the `.DS_Store` files could the cause of the size difference. I don't own a Mac to test with, so I can't check whether the `__MACOSX` folder might cause problems for the theme system. It is possible to create .zip files on macOS without this folder:
https://stackoverflow.com/a/13665552
Give that a try and see if it makes the Mac-created .zip themes start working.

R0dri, 2019-04-05:
> Are you sure that the folder structure of the working and non-working .zip files are exactly the same?
 Regarding anything I could do about it... I believe so because after decompressing and compressing same folder without even enter in it would cause it to stop working. Cant see any subfolder or unchanged file structure.

> I know that .zip files created on macOS will have a __MACOSX folder added. This, and the .DS_Store files could the cause of the size difference.
I think you might be on to something. Will test right now

ps: sorry for being idle, my notifications were turned off for some reason

R0dri, 2019-04-08:
forgot to mention, managed to make it work by compressing the same folder on a windows VM.

per1234, 2019-04-11:
@R0dri please attach the non-working theme .zip file in a reply here.

R0dri, 2019-04-18:
[DarkArduinomod1.zip](https://github.com/arduino/Arduino/files/3094359/DarkArduinomod1.zip)
 
I just uncompressed a working zip, and recompressed on my Mac. This is the output file, and it is not recognized by Arduino again.

facchinm, 2019-04-18:
@R0dri 
the problem seems to be the fact that MacOs creates a folder on extraction, even if the original archive didn't have one. A working zip contains `theme.properties` file in the first level, while your zip has two folder on that level, once called `DarkArduinoMod1` and the usual `__MACOSX` which always bothers me :slightly_smiling_face: 

To repackage the zip correctly, open a terminal, `cd` into the theme directory and then execute `zip -r ../DarkArduinomod2.zip *` . 
The resulting zip wont have extra folders and will be recognized by the IDE.

R0dri, 2019-04-18:
Wow, that was it. I can't believe it. I thought I tried every possible way to zip the folder from terminal using every option. I even tried to see all files/folders from the zip file without uncompressing it and couldn't find __MACOSX anywhere. how do you see that it is there?

facchinm, 2019-04-18:
>  how do you see that it is there?
Using Linux :smile:

R0dri, 2019-04-18:
haha, good to know. If you don't mind me asking. Could you explain to me the command? and why it worked?

facchinm, 2019-04-18:
Sure! The command calls `zip` to recursively (`-r`, so subfolders are in) compress all files (`*`) and output them to `../DarkArduinomod2.zip` . Since you are in the DarkArduinomod2 folder, the first level of the directory becomes the first level of the zip, and the IDE is happy

R0dri, 2019-04-18:
haha interesting. So it wouldn't work the same if I do it outside the folder pointing at DarkArduinomod2/* . That would get me in the same trouble, right?

facchinm, 2019-04-18:
Exactly, it would have a first level `DarkArduinomod2` folder (as in Mac compressor, but without the __MACOS` folder)