arduino/Arduino #8703: [Core] Enhancement to micros() and millis() to be able to change prescaler.

GewoonGijs, 2019-03-25:
Made some adjustments to wiring.c to mae micros() and millis() resilient to changes of the prescaler

https://github.com/GewoonGijs/platformio-pkg-framework-arduinoavr

GewoonGijs, 2019-03-27:
I changed the calculations for micros() and millis() to be dependant on the actual state of the Timer0 configuration registers. This makes it psossibel to change the prescaler and still have correct working timing functions.

Changes are only made in wiring.c

xtofl, 2020-01-01:
Encountered the same issue when playing with the Prescaler: `delay()` stops working properly when changing the prescaler (on a Gemma board, with attiny85).  Since `delay` is implemented in terms of `micros()`, and `micros()` in terms of `clockCyclesPerMicrosecond`, the problem does not seem to be in `wiring.c`, but rather in

```
#define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )
```
Which is defined in "Arduino.h".

I'm wondering if there is a way to somehow add a user faced prescaler function (e.g. through a macro :( ) that also changes the `clockCycles....` and possibly other time related functions to a soft-coded implementation, e.g.

```
// client code example
#define ARDUINO_I_WANT_TO_USE_PRESCALER
#include <Arduino.h>

void setup() {
   setPrescalerBits(1,1,1,1); // this would give a decent error when the `USE_PRESCALER` is not defined
}
```

The ticks-functions can be conditionally defined:

```
#ifdef ARDUINO_I_WANT_TO_USE_PRESCALER
inline long clockCyclesPerMicrosecond() {
  return BASE_CLOCK * prescalerFactor();
}
#else
#define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )
#endif
```

matthijskooijman, 2020-03-29:
This was re-reported at https://github.com/arduino/ArduinoCore-avr/issues/79 where it is now more appropriate, so I'm closing this one as duplicate.

closed 2020-03-29 by matthijskooijman