arduino/Arduino #8707: [Library Manager] Library ESP-DASH not getting updated in your DB

ayushsharma82, 2019-03-27:
Hi,
My Library ESP-DASH (https://github.com/ayushsharma82/ESP-DASH) is not getting updated in library manager.

Can anybody tell me if this is a error or a mistake from my end?

Regards,
Ayush Sharma

per1234, 2019-03-27:
The reason your 2.1.0 - v2.1.2 releases weren't picked up by the Library Manager indexer job is that you didn't update the `version` field in library.properties before creating tags or releases. The Library Manager indexer will only pick up tags that contain a new `version` value in their library.properties.

I notice that you did do this for the v2.1.3 release:
https://github.com/ayushsharma82/ESP-DASH/blob/v2.1.3/library.properties#L2
I also notice that you changed the library name from `ESP-DASH` to `ESP DASH` in that release:
https://github.com/ayushsharma82/ESP-DASH/commit/7816f6599705fb89ce3a631603e7cfccc4fdfcc6
I wasn't aware that would block the release from being picked up by the indexer job, but that's the only issue I notice with that release.

You need to submit a request here in the issue tracker if you want to change the library name.

ayushsharma82, 2019-03-27:
Oh ok. So if i revert the name of library to previous one, Then will it work?

per1234, 2019-03-27:
I believe so. cmaglie can provide a definitive answer when they get a chance to take a look at this issue report. However, if you do want to stick with the new library name, cmaglie can do that for you. You just have to request that to be done here.

ayushsharma82, 2019-03-27:
It's Fine. I will first try with renaming back to ESP-DASH.

ayushsharma82, 2019-03-27:
Yup. Renaming back to original name got it updated in Arduino Library Manager. Thanks for your support!

closed 2019-03-27 by ayushsharma82