arduino/Arduino #8710: [Library manager] Add my library.

TomasRoj, 2019-03-27:
Hey guys,

I just made simple Arduino library for beginners. They can just call a function and dont must write hard code. There are only 6 methods there right now but more will be coming. I also made docs for it:  www.TomasRoj.github.io/BasicsLibrary . Name of the library is BasicsLibrary like basics examples for beginners (hope you understand xd). I did a lot of work. Its not just weekend project. I want to work on it until there will be maximum examples for beginners as we can do. I also hope if someone can help me by creating this project. Together, we can build library that will help beginners to start and they will not escape from this community!  If you are interested write me please on this email: rojtomas@email.cz

Source code: https://github.com/TomasRoj/BasicsLibrary

Cheers,
Tomas Roj

per1234, 2019-03-27:
There are a couple of things that will need to be resolved before your library can be added to the Library Manager index:

The library must be in the root of the repository. It is currently in the BasicsLibrary subfolder of the repository, which is not allowed.

After moving the library, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing these things, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

TomasRoj, 2019-03-28:
Hi,
I changed thing waht you want. I think now it si ready for publish!

TomasRoj, 2019-03-28:
Hey,

I see that you removed waiting for feedback tag but the library isnt there yet. Is there some problem or something?

per1234, 2019-03-28:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to take care of all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that.

So there is no other action required on your part other than to be patient. Thanks!

cmaglie, 2019-03-29:
Done!

closed 2019-03-29 by cmaglie