arduino/Arduino #8716: Add more explanation about --board parameters at manpage.adoc

AnnyCaroline, 2019-03-28:
Hi everyone,

I'm using [manpage.adoc](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc) as my guide to use Arduino IDE via command line.

It would be awesome if the guide provides more information about the parameters used with `--board` option, for example, a list of the possible parameters and example of usage.

per1234, 2019-03-28:
>a list of the possible parameters

There's no way to list the possible parameters because anyone who makes a board definition can add any parameters they like. There is no standardization imposed on the parameters.

This is somewhat related to: https://github.com/arduino/arduino-cli/issues/138
In my reply there, I described how you could determine the parameters from looking at the boards.txt entry for the target board, or how you can more easily get the complete FQBN via the Arduino IDE's GUI:
https://github.com/arduino/arduino-cli/issues/138#issuecomment-459169051

---
>and example of usage.

There is one already here: https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc#examples
>Compile and upload a sketch to an Arduino Nano, with an Atmega168 CPU, connected on port /dev/ttyACM0:
    ```
    arduino --board arduino:avr:nano:cpu=atmega168 --port /dev/ttyACM0 --upload /path/to/sketch/sketch.ino
    ```

AnnyCaroline, 2019-03-28:
Really Thanks. This helped me a lot :)

closed 2019-03-28 by AnnyCaroline