arduino/Arduino #8717: [Library Manager] Add support for Sensirion SPS30

winkj, 2019-03-28:
Hi there,

Could you add https://github.com/Sensirion/arduino-sps to the Library Manager?

It's a driver for Sensirion's SPS30 particulate matter sensor (https://www.sensirion.com/sps30).

Let me know if you have any questions.
Best, Johannes

per1234, 2019-03-28:
Hi. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your library to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

winkj, 2019-03-28:
Makes sense, missed that change!

I'll update and resubmit, thanks for the explanation

winkj, 2019-03-29:
I renamed the library in library properties - from arduino-sps (old) to sensirion-sps (new) - and tagged a new release.

Let me know if there's anything else that needs to be changed!
Thanks

winkj, 2019-04-10:
I will close this because we discovered an issue on non-AVR platforms; will resubmit when addressed

winkj, 2019-04-12:
Ensured support for AVR, SAMD, ESP8266 and ES32

cmaglie, 2019-04-23:
done

closed 2019-04-23 by cmaglie