arduino/Arduino #8719: [Library Manager]Please add Grove - I2C High Accuracy Temp&Humi Sensor(SHT35) v1.0.0 to Library Manager.

Jenkinlu001, 2019-03-29:
Github:https://github.com/Seeed-Studio/Seeed_SHT35
Release:https://github.com/Seeed-Studio/Seeed_SHT35/releases/tag/v1.0.0

Thanks

per1234, 2019-03-29:
Hi. The `name` value in library.properties for this library (`Grove - I2C High Accuracy Temp&Humi Sensor(SHT35)`) is not compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>Library names must contain only basic letters (A-Z or a-z) and numbers (0-9), spaces (), underscores (_), dots (.) and dashes (-). It cannot start or end with a space, and also it cannot start with a number.

Please change the `name` value to something compliant with the specification and then either update the `version` value in library.properties and create a new tag/release or else redo the existing tag/release.

After you have done this, please comment here so we'll know we can proceed.

Jenkinlu001, 2019-03-29:
I'll update it and submit it again, thanks

Jenkinlu001, 2019-03-29:
This is the latest:
rename:Grove - I2C High Accuracy Temp_Humi Sensor SHT35
Github:https://github.com/Seeed-Studio/Seeed_SHT35
Release:https://github.com/Seeed-Studio/Seeed_SHT35/releases/tag/v1.0.0

Thanks

cmaglie, 2019-04-12:
done!

closed 2019-04-12 by cmaglie