arduino/Arduino #8721: [Library Manager]Please add Grove - I2C Thermocouple Amplifier MCP9600 v1.0.0 to Library Manager.

Jenkinlu001, 2019-03-29:
Github:https://github.com/Seeed-Studio/Seeed_MCP9600
Release:https://github.com/Seeed-Studio/Seeed_MCP9600/releases/tag/v1.0.0

Thanks

per1234, 2019-03-29:
Hi. I see you updated the `name` value to remove the disallowed characters, but in the latest tag, the `name` value is still non-compliant:
https://github.com/Seeed-Studio/Seeed_MCP9600/blob/v1.0.0/library.properties#L1

Please either update the `version` value in library.properties and then create a new tag/release, or else redo the existing release.

Once you have done this, please comment here so we'll know we can proceed.

Jenkinlu001, 2019-03-29:
update Release: https://github.com/Seeed-Studio/Seeed_MCP9600/releases/tag/v1.0.0

cmaglie, 2019-04-12:
done!

closed 2019-04-12 by cmaglie