arduino/Arduino #8729: eu_dst.h wrong calculation od Daylight Saving Time

vprheli, 2019-03-29:
Calculation use wrong conditions. Please correct
        if              (month == MARCH) {
            if (mday < d)
                return 0;
            if (mday > d)
                return ONE_HOUR;
            if (hour < 2)
                return 0;
            return ONE_HOUR;
        }
        if              (mday < d)
                            return ONE_HOUR;
        if              (mday > d)
                            return 0;
        if              (hour < 2)
                            return ONE_HOUR;
                        return 0; 

And check also usa_dst.h if there isn't similar error
Vladimir

facchinm, 2019-04-01:
Hi @vprheli ,
which file/toolchain are you referring to?

Robotto, 2019-04-24:
It's the from the AVR libc toolchain.
more info here: http://avr.2057.n7.nabble.com/bug-44327-eu-dst-is-broken-td21561.html

I'd vote against implementing the proposed fix from the above link, since it's ungraspable to ever fix again. But the link serves the purpose of confirming the bug.

Robotto, 2019-04-25:
It seems i've fixed it, using most of the original code... but.. I'm not sure if eu_dst.h is included in any of the arduino git repos, so i'm not sure what to do now... 
If @someone could point me in the direction of the file, id be happy to do a pull request mentioning this issue...