arduino/Arduino #8731: [Library Manager] Library GSL1680 not updating

ESTBLC, 2019-03-29:
Hi,

I have removed some old unusable released and now my new releases don't seem to be fetched by the lib manager

What should I do ?

per1234, 2019-03-30:
The Library Manager index contains entries for your releases 1.0.0, 1.0.1, and 1.1.0. The only releases your repository currently contains are 1.0.0 and 1.1.0. The 1.1.0 release you made 4 hours ago was indeed fetched. Perhaps it had not yet been fetched at the time you submitted this report. The [FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) implies this would happen within an hour and I see you submitted this issue over two hours after you created the 1.1.0 release. Perhaps now that there are so many libraries in the index, it takes over an hour. Alternately, maybe I'm misinterpreting the FAQ (which doesn't actually say it will be done in an hour).

I assume the 1.0.1 release was the one you removed. The reason this removal has not been reflected in the Arduino Library Manager index is explained here:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#sorry-i-did-something-wrong-how-can-i-change-or-unpublish-an-already-published-library
>Modified or Deleted library versions are processed every Sunday.

So it appears to me that everything is working well enough with your repository now. I'll go ahead and close the issue, but we should consider making the FAQ more clear if it does take >1 hour for the releases to show up in the index.

closed 2019-03-30 by per1234

ESTBLC, 2019-03-30:
In did everything is working well, I have not noticed the line you mention in the FAQ.

I have add something in the FAQ to make the >1 hour obvious see :
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ/_compare/184889a7c2aa26b148d4c6e085cece2d9c794936

Thanks for the support