arduino/Arduino #8732: [Library Manager] Add `Ephemeris` to Library Manager

MarScaper, 2019-03-30:
https://github.com/MarScaper/ephemeris

Simple C++ library allowing to compute planet coordinates (equatorial and horizontal) with Arduino (>=Mega).

**Features:**
- Compute equatorial coordinates (RA/Dec), horizontal coordinates (Alt/Az), rise, set, distance (in A.U), and apparent diameter (in arcminutes) for Sun, Mercury, Venus, Earth's Moon, Mars, Jupiter, Saturn, Uranus, Neptune.
- Uses VSOP87 for planets.
- Uses ELP2000 for the Moon.
- Code optimized for single-precision calculations with Arduino.

**Goodies**
- Transformations from equatorial coordinates (RA/Dec) to horizontal coordinates (Alt/Az).
- Transformations from horizontal coordinates (Alt/Az) to equatorial coordinates (RA/Dec).
- Rise and set for custom equatorial coordinates.
- Time zone conversion.
- Conversion from equatorial coordinates for a specified equinox (Ex:J2000) to apparent equatorial coordinates (JNow).

**Limitations**
Due to VSOP87 and ELP2000 implementation, code needs too much flash memory for entry level Arduinos (Uno, etc).

per1234, 2019-03-30:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

MarScaper, 2019-03-30:
Thanks. Release tag is done...
https://github.com/MarScaper/ephemeris/releases

cmaglie, 2019-04-12:
done!

closed 2019-04-12 by cmaglie