arduino/Arduino #8734: [Feature Request] Make Wire library methods virtual

simonnilsson, 2019-03-30:
I suggest making the public methods of the TwoWire class virtual in order to allow polymorphism when subclassing it.

**Use case**: Have another library that utilizes the Wire library and takes a refrence to a TwoWire instance in constructor. Want to use it on pins without hardware I2C (or other reason for not wanting to use the standard Wire implementation). So I want to pass another class instance that inherits from TwoWire and implements the same methods (but uses software I2C). This does not work with current implementation because of early binding (the base class functions will be called instead of the ones in subclass).

facchinm, 2019-04-02:
Hi @simonnilsson , would you mind moving this proposal to https://github.com/arduino/ArduinoCore-API repo? Thanks!

simonnilsson, 2019-04-02:
Move to: https://github.com/arduino/ArduinoCore-API/issues/31

closed 2019-04-02 by simonnilsson