arduino/Arduino #8741: Adding a dysfunct proxy creates strange errors and does not produce useful error messages

holgerlembke, 2019-04-01:
As said. Just try it, add a not working proxy (I had "localhost" and port 8118).

It creates all sorts of java errors instead of useful error messages.

facchinm, 2019-04-02:
Hi @holgerlembke ,
which operating system are you using? Can you past the "all sorts of java errors" ? I'm only getting 
```
 Error downloading https://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:333)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:307)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:285)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:146)
	... 1 more
Caused by: java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient$1.run(HttpClient.java:515)
	at sun.net.www.http.HttpClient$1.run(HttpClient.java:513)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.http.HttpClient.privilegedOpenServer(HttpClient.java:512)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:553)
	at sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:264)
	at sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:367)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:191)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1199)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:177)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:162)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:181)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more
```
on Linux (which is totally expected and very informative to understand what is going wrong)

holgerlembke, 2019-04-02:
I'm using windows, error looks like the same. And the IDE seems to stop responding for some seconds (more than 5 secs...).

```
Der Sketch verwendet 2258475 Bytes (71%) des Programmspeicherplatzes. Das Maximum sind 3145728 Bytes.
Globale Variablen verwenden 51512 Bytes (15%) des dynamischen Speichers, 276168 Bytes für lokale Variablen verbleiben. Das Maximum sind 327680 Bytes.
 Fehler beim Herunterladen von https://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Fehler beim Herunterladen von https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Fehler beim Herunterladen von https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:333)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:307)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:285)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:146)
	... 1 more
Caused by: java.net.ConnectException: Connection refused: connect
	at java.net.DualStackPlainSocketImpl.waitForConnect(Native Method)
	at java.net.DualStackPlainSocketImpl.socketConnect(DualStackPlainSocketImpl.java:85)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient$1.run(HttpClient.java:515)
	at sun.net.www.http.HttpClient$1.run(HttpClient.java:513)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.net.www.http.HttpClient.privilegedOpenServer(HttpClient.java:512)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:553)
	at sun.net.www.protocol.https.HttpsClient.<init>(HttpsClient.java:264)
	at sun.net.www.protocol.https.HttpsClient.New(HttpsClient.java:367)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.getNewHttpClient(AbstractDelegateHttpsURLConnection.java:191)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1199)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
	at sun.net.www.protocol.https.AbstractDelegateHttpsURLConnection.connect(AbstractDelegateHttpsURLConnection.java:177)
	at sun.net.www.protocol.https.HttpsURLConnectionImpl.connect(HttpsURLConnectionImpl.java:162)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:181)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more
```

We can close this. I just thought that the desired target audience of Arduino can't connect this messages to a dysfunct proxy and that exceptions should be handled and replaced this reasonable help messages.

facchinm, 2019-04-02:
Proxies are very nasty stuff, and the stacktrace helps us getting some insight of what is going on. It may be jargon (from the user point of view) but it gives all the information needed for someone help the user fixing the issue :wink:

closed 2019-04-02 by facchinm