arduino/Arduino #8743: [Library Manager] Please add libraries

hideakitai, 2019-04-02:
Hello, I created some libraries. Could you add following to a list of libraries ?

ArtNet
https://github.com/hideakitai/ArtNet

Thanks in advance.

per1234, 2019-04-03:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

hideakitai, 2019-04-06:
Sorry, I missed it... I have created release, please check it.

cmaglie, 2019-04-12:
done

closed 2019-04-12 by cmaglie