arduino/Arduino #8751: Converting float to int

mariusvds, 2019-04-04:
Hello world,

This week i found a bug, converting float to int delivers strange values, can't find out why.

This is the code I wrote:

float answer = 0.0;
float i = 0.0;

void setup()
{
  Serial.begin(115200);
  Serial.println("Hello World !!!");
}

void loop()
{
  while(true)
  {
    answer = pow(5.0, i);
    Serial.print("Answer: ");
    Serial.println(answer);
    if (i >= 4.0){i = 0.0;}; //preventing overflow
    int b = static_cast<int>(answer);//converting to int
    Serial.print("B: ");
    Serial.println(b);
    delay(500);
    i += 1.0;
  }
}

The output is:
Hello World !!!
Answer: 1.00
B: 1
Answer: 5.00
B: 5
Answer: 25.00
B: 24
Answer: 125.00
B: 124

Somebody who can find out what goes wrong?

Thanks

mikaelpatel, 2019-04-04:
I think you are asking this question in the wrong place. Try the forum or stackexchange instead. 

BW: looks like you are truncating the float to integer. Might what to try rounding it instead.

per1234, 2019-04-05:
The issue will be apparent if you print `answer` with more decimal places:
```c++
Serial.println(answer, 6);
```
We have already documented the inexact nature of float [in the Arduino Language Reference](https://www.arduino.cc/reference/en/language/variables/data-types/float/) so I don't think any action is required on this.

closed 2019-04-05 by per1234

PaulStoffregen, 2019-04-05:
@per1234 - Perhaps the documentation could specifically mention the need to add 0.5 for converting to nearest integer?  Or maybe the round() function should be documented?

per1234, 2019-04-05:
>Perhaps the documentation could specifically mention the need to add 0.5 for converting to nearest integer?

@PaulStoffregen good suggestion! I hadn't considered the bigger picture behind this issue. The truncation behavior when converting from floating point to integer types is a frequent cause of confusion for beginners. Since I already have a lot of outstanding PRs in the Language Reference repo that are requiring me to frequently resolve merge conflicts, I'm holding off on any submitting any new PRs to that repo for now. So I created a dedicated issue for your suggestion instead:
https://github.com/arduino/reference-en/issues/579

> Or maybe the round() function should be documented?

We do have an existing issue report for this:
https://github.com/arduino/reference-en/issues/522
Now that I've been [informed](https://github.com/arduino/reference-en/issues/476#issuecomment-461491446) that .ino files are written in Arduino Language, rather than C++ as I had previously thought, I am warming to the idea of implementing that request, since the Arduino Language Reference is the only definitive documentation of the Arduino Language.