arduino/Arduino #8753: Preferences: http://arduino.esp8266.com/stable/package_esp8266com_index.json

akshaymali046, 2019-04-05:
Error:
Error downloading https://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:333)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:307)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:285)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:146)
	... 1 more
Caused by: java.nio.file.FileSystemException: C:\Users\Patil\AppData\Local\Arduino15\package_index.json.tmp: The process cannot access the file because it is being used by another process.

	at sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:86)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:102)
	at sun.nio.fs.WindowsFileSystemProvider.implDelete(WindowsFileSystemProvider.java:269)
	at sun.nio.fs.AbstractFileSystemProvider.delete(AbstractFileSystemProvider.java:103)
	at java.nio.file.Files.delete(Files.java:1126)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:156)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more
 Error downloading https://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:333)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:307)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:285)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:146)
	... 1 more
Caused by: java.nio.file.FileSystemException: C:\Users\Patil\AppData\Local\Arduino15\package_index.json.tmp: The process cannot access the file because it is being used by another process.

	at sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:86)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:102)
	at sun.nio.fs.WindowsFileSystemProvider.implDelete(WindowsFileSystemProvider.java:269)
	at sun.nio.fs.AbstractFileSystemProvider.delete(AbstractFileSystemProvider.java:103)
	at java.nio.file.Files.delete(Files.java:1126)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:156)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more

facchinm, 2019-04-05:
It looks related with aggressive antivirus or something like this, which is holding a lock on `C:\Users\Patil\AppData\Local\Arduino15\package_index.json.tmp` . Try pausing your antivirus while opening Board Manager to check if it gets fixed.

juandi22, 2019-07-13:
go into: C:\ProgramData\MATLAB\SupportPackages\R2018a\3P.instrset\arduinoide.instrset\idepkgs
 and delete package files

per1234, 2019-09-16:
@akshaymali046 we're still waiting for you to let us know if disabling the antivirus resolved this issue for you (or if there was some other resolution you found). Please take a minute to provide us with an update.

misbagas, 2019-11-21:
how to do in OS mac

per1234, 2020-08-04:
Closing due to lack of response.

@akshaymali046 if you'll provide the requested information I'll be happy to re-open this issue.

closed 2020-08-04 by per1234