arduino/Arduino #8754: Editor: Hilite all occurences of selected word

s-stexx, 2019-04-05:
I would really like to edit all of my sketches in Arduino. There is one major feature, that I miss, when using the built-in editor: hilite all occurences, of the text I just selected.

It doesn't need to be just variables, that are hilited - it can be just anything (even comments) that have the same chars in it.

Even nearly all modern editors offer this feature, I like BB Edit on Mac most, because it hilites every occurence.

s-stexx, 2019-04-05:
Here are 2 examples:
1. I selected the variable "hours" in the last line:
[image]
2. I selected the string "print" in the last line:
[image]

facchinm, 2019-04-08:
Looks similar to this https://github.com/bobbylight/RSyntaxTextArea/issues/88 . A possible implementation is here too https://github.com/arduino/Arduino/pull/8012 (for find/replace functionality only though)