arduino/Arduino #8758: [Library Manager] add VincentLim/TimerFour

photodude, 2019-04-08:
Please consider adding https://github.com/VincentLim/TimerFive to the library manager index

This is a modification of the [PaulStoffregen/TimerThree](https://github.com/PaulStoffregen/TimerThree) Library to provide the same functionality for modifying Timer Four on supported boards

per1234, 2019-04-08:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

photodude, 2019-04-10:
@per1234, this one should also be ready to go.

cmaglie, 2019-04-12:
done!

closed 2019-04-12 by cmaglie

photodude, 2019-04-12:
@per1234 @cmaglie Thank you