arduino/Arduino #8763: [Library Manager] Add requet to Library Manager

andhieSetyabudi, 2019-04-09:
Greetings.
Please add basic statistic to the library manager. 
Here's the github link:
https://github.com/andhieSetyabudi/filter-library/tree/master/basic_statistic
Thank you very much!

per1234, 2019-04-09:
In order to be included in the Arduino Library Manager index, your library must be located in the root of the repository. There is a library in the root of the repository, but the link you provided points to the other library in the basic_statistic subfolder. It seems these are two different versions of the same library. This is bound to cause confusion and problems with the administration of your project. I recommend that you structure the repository so it contains only a single copy of the library, which is in the root of the repository.

Once that's done, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

per1234, 2019-07-26:
Since there has been no response after >3.5 months, I'm going to go ahead and close this. @andhieSetyabudi if you ever get your library compliant with the Library Manager requirements please comment here and I'll re-open the request.

closed 2019-07-26 by per1234