arduino/Arduino #8766: Arduino code no longer works after a few restarts and unplugs and resets (Read)

xcakepiggie, 2019-04-09:
Hi,

        I have tried to build a simple keyboard with this code: (not written by me)
`
/*
 *   LastUpdate:2017/9/18
 *   Version:1.0
 *   Anothor:SASA2981
 *   Comment:
 *     とりあえずZXが入力できるように作成したもの
*/

#include <HID-Project.h>
#include <TimerOne.h>

// ピン番号
#define LLPIN 4
#define LPIN 3
#define RPIN 2

// 一つ前の入力状態
int oldInputData;

void setup() {
  
  // シリアル通信開始
  Serial.begin(9600);

  // キーボード使用
  NKROKeyboard.begin();
  oldInputData = 0;

    // ピン設定
  pinMode(RPIN, INPUT_PULLUP);
  pinMode(LPIN, INPUT_PULLUP);
  pinMode(LLPIN, INPUT_PULLUP);
  
  // タイマー割り込み
  Timer1.initialize();
  
  // 1msごとに割り込み処理
  Timer1.attachInterrupt(DataSend,1000);
  
  oldInputData = 0;
}

void loop() {

}

void DataSend(){
  
  // 現在の入力状態
  int InputData = 0;
  
  // スイッチ状態取得
  InputData |= (!digitalRead(RPIN) << RPIN);
  InputData |= (!digitalRead(LPIN) << LPIN);
  InputData |= (!digitalRead(LLPIN) << LLPIN);
  
  // 前回と異なれば入力更新
  if(InputData != oldInputData){

    // 入力状態更新
  UpdateInput(InputData & (1 << RPIN), 'x');
  UpdateInput(InputData & (1 << LPIN), 'z');
  UpdateInput(InputData & (1 << LLPIN), 'z');

    // PC側に送信
    NKROKeyboard.send();
  }
  
  // 前回のバッファにコピー
  oldInputData = InputData;
}

// 入力状態更新
void UpdateInput(bool inputstate, char key){

  if(inputstate){
    // 入力
    NKROKeyboard.add(key);
  }else{
    // 解除
    NKROKeyboard.remove(key);
  }
}
`

        It didn't work at first until I restarted my pc. After a restart, it worked fine and I uploaded THE SAME CODE again and now all of a sudden, it doesn't work (no keys inputted but Arduino can still read HIGH and LOW)

        And this time, a restart didn't work. It isn't a power issue.

per1234, 2019-04-09:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there. If we do end up determining that your issue is caused by a bug in some of Arduino's code, I'll be happy to re-open this issue.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2019-04-09 by per1234

xcakepiggie, 2019-04-10:
> This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there. If we [...]

the website is really slow and I can't make an account because it is too slow and has some html issues

per1234, 2019-04-11:
@xcakepiggie I'm sorry for any inconvenience. Very rarely the Arduino Forum will be running a bit slow or having issues, but it's always fixed within a matter of hours. I'm sure you'll be able to get on if you try again.

I have been using the forum constantly over the last days and have not had any problems.

If you try again and are still having troubles, it might be an issue on your system. If you have any ad blocker or noscript types of browser extensions installed, try disabling them.