arduino/Arduino #8777: libraries with very old dates (year 1979)

jsmith173, 2019-04-12:
Hi,

My library files (for example Wifi101) have very old dates (11/30/1979/00:00)

Arduino: 1.8.8
Op: Windows 10

facchinm, 2019-04-15:
@cmaglie could it be due to a misconfigured time on jenkins side at some point? Should we rebuild the offending archives (or all of them) ?

cmaglie, 2019-04-23:
not sure why this is happening, the full collection is rebuilt every week, so I doub't it's a misconfigured time, will chack this out tomorrow

per1234, 2019-07-01:
I just tried again and the problem no longer occurs for me, so I think it might have been fixed. @jsmith173 is it fixed for you too?

cmaglie, 2019-07-03:
Yes it should be fixed now.

BTW, I'ts worth noting that git doesn't store timestamps, so the generated zip will have the timestamp of the moment it is created (or regenerated if this will be necessary for some reason in the future), so the timestamp of the file may change and consequently the CRC.

closed 2019-07-03 by cmaglie