arduino/Arduino #8779: Arduino IDE 1.8.9 does not install correctly on

CedricMi, 2019-04-12:
Arduino IDE 1.8.9 does not install correctly on Debian distributions, or at least I tried: Ubuntu 18.04.2 and Pengwin.
With sudo ./install.sh one ends up with:

```
Adding desktop shortcut and menu item for Arduino IDE...
rm: cannot remove '/usr/local/bin/arduino': No such file or directory
Removing symlink failed. Hope that's OK. If not then rerun as root with sudo.

rm: cannot remove '/usr/local/bin/arduino': No such file or directory
Removing symlink failed. Hope that's OK. If not then rerun as root with sudo.

Done!
```

The files are not copied correctly to whatever directory (/home/Arduino, /home/.arduino15, /usr/local/bin/arduino ?) they are supposed to be copied.

Please fix.

Also when launched from a terminal window

```
Picked up JAVA_TOOL_OPTIONS:
Gtk-Message: timestamp: Failed to load module "Canberra-gtk-module"
```

facchinm, 2019-04-12:
Do you have the folder `/usr/local/bin/` ? In that case, launching with sudo should create the right symlinks (there). It first tries to remove but continues if it doesn't succeed. The GTK warning is originated by java and there's very little we can do about it.

CedricMi, 2019-04-12:
The folder /usr/local/bin/ is present.
Inside, there is arduino, but it isn't a folder, it is a file!

facchinm, 2019-04-12:
It's fine, since it's a symlink to the executable you extracted in your local directory you can now launch `arduino` from the commandline. If you prefer doing things graphically, the install script should also have created an icon on the desktop.

CedricMi, 2019-04-12:
- The icon on the desktop doesn't work: There was an error launching the application.
https://imgur.com/8BdHBbP
- If I remove the extracted archive, nothing works, from CL or GUI. That can't be right. Have you tried to install it?

facchinm, 2019-04-12:
> If I remove the extracted archive, nothing works, from CL or GUI. That can't be right. Have you tried to install it?
This is expected, the installer just creates links to the extracted archive
> The icon on the desktop doesn't work: There was an error launching the application.
No idea what is happening here, but may be related to the deletion of the extracted archive

The installer just creates icons, links and files association. The real "installation" is self contained in the extracted archive (which should not be deleted)

calocedrus, 2019-04-17:
I have the exact same issues as CedricMi (ubuntu 18.04), following the exact same installation procedure. I didn't worry about the warning in the terminal though.
I have tried the installation in a local folder in /home/user as well as in /opt (with sudo rights), all gave:
```
Adding desktop shortcut and menu item for Arduino IDE...
rm: cannot remove '/usr/local/bin/arduino': No such file or directory
Removing symlink failed. Hope that's OK. If not then rerun as root with sudo.

rm: cannot remove '/usr/local/bin/arduino': No such file or directory
Removing symlink failed. Hope that's OK. If not then rerun as root with sudo.

Done!
```
The desktop icon doesn't look like an icon and has the name `arduino-arduinoide.desktop`

@facchinm 
>> The icon on the desktop doesn't work: There was an error launching the application.

>No idea what is happening here, but may be related to the deletion of the extracted archive

This could have been a reason but I (obviously :-) ) didn't delete the extracted archive.

This may be a bug in the installation file / procedure.

calocedrus, 2019-04-17:
Ok well, had a look in the `install.sh`, the `cannot remove /usr/local.../` message is totally normal on a system without prior arduino installation, since the install script runs an uninstall procedure which tries to remove a symlink which doesn't exist. So this can be ignored.

Remains the weird desktop icon which doesn't start.

per1234, 2019-04-17:
Note that the spurious "rm: cannot remove '/usr/local/bin/arduino'" warnings on the first install are removed by https://github.com/arduino/Arduino/pull/8781

calocedrus, 2019-04-17:
And finally a way to get the icon work: 

`sudo chown /path/to/icon/arduino-arduinoide.desktop yourusername` 

Double click and trust the launch.
Here you go :+1: 
All is well now (but there may be some tweaks to bring in the script to avoid this).

facchinm, 2019-04-18:
I've been finally able to test on a clean Ubuntu 18.10; the user installation is fine (just reports warnings on not being able to symlink to `/usr/local/bin`. Sudo installation though has the icon problem; to solve it I just pushed https://github.com/arduino/Arduino/commit/c3675a8befbc9d76850a0e986e6b0da0b8b43084 to fix this automatically. I think the issue can be closed as solved (in master).

closed 2019-04-18 by facchinm

nhphuong91, 2019-05-22:
> And finally a way to get the icon to work:
> 
> `sudo chown /path/to/icon/arduino-arduinoide.desktop yourusername`
> 
> Double click and trust the launch.
> Here you go +1
> All is well now (but there may be some tweaks to bring in the script to avoid this).

I think it should be:
`sudo chown YourUserName /path/to/icon/arduino-arduinoide.desktop`
otherwise chown would give error `invalid user: ‘/path/to/icon/arduino-arduinoide.desktop’`

KikiCS, 2019-05-31:
Had the same problem installing today on RaspberryPi with Ubuntu Mate, solved with @nhphuong91 command:
`sudo chown UserName /home/UserName/Desktop/arduino-arduinoide.desktop`

Basilisvirus, 2019-08-13:
Hello, as mentioned in https://github.com/arduino/Arduino/commit/c3675a8befbc9d76850a0e986e6b0da0b8b43084 , the problem should have been solved right?
I installed the arduino ide today, and i still get `Adding symlink failed. Hope that's OK. If not then rerun as root with sudo.`

The shortcut on the desktop is broken, i tried `sudo chown UserName /home/UserName/Desktop/arduino-arduinoide.desktop` , which fixed the image on the shortcut, but it is still broken. 

I then tried to re-install it without sudo, and it poped the error several times: `touch: cannot touch ... `
and at the end, everything is broken again. I did not delete the unziped file, and i only tried to install using `sudo ./install.sh` and the, obviously `./install.sh`

ArghyaChatterjee, 2019-08-19:
I had the same issue while running on nvidia jetson nano with ubuntu-18.04 LTS and arduino-1.8.9. I followed the follwing command and worked for me.

```sudo chown UserName /home/username/Desktop/arduino-arduinoide.desktop```

After I tried to run the arduino ide, it pops up with a message if I trust this software, just accept it and you are ready to go.

facchinm, 2019-08-19:
The fix has been included in master branch and will be available starting from IDE 1.8.10, so 1.8.9 is still expected to fail (and `chown` must be applied manually)

zhangshaohu, 2019-09-09:
I installed the older version. It works. Version 1.8.4

sevenp240, 2019-11-27:
Having this issue on version 1.8.10 on Arm64. Chown does not fix it.

dmcquestion, 2020-08-04:
Having these issues on version 1.8.13 on ARM64 running on a Raspberry Pi4 Model B w/ 4G RAM

Download and Extract arduino-1.8.13-linuxaarch64.tar.xz
cd /home/pi/Downloads/arduino-1.8.13
sudo ./install.sh
Adding desktop shortcut, menu ..<blah><blah>...
touch: cannot touch '/root/.config/mimeapps.list': No such file or directory /usr/bin/xdg-mime: 848: /usr/bin/xdg-mime: cannot create /root/.config/mimeapps.list.new: Directory nonexistent
 done!

I get the desktop icon 'arduino-arduinoide.desktop'

I execute the chown command:
sudo chown pi /home/pi/Desktop/arduino-arduinoide.desktop

When I click on the desktop icon or go to Menu > Programming > Arduino IDE, I get the hour glass on my cursor for about 30 seconds and nothing.

This is my 9th or 10th installation attempt.  I have tried multiple with the 64-bit.  I have tried a couple with the 32-bit.  I have tried using other instructions that have you move the arduino-1.8.13 to the /opt directory and try the installation from there.  I have tried the installation with ./install.sh and sudo ./install.sh.  All failed with various different errors and icon results. 

I am open for suggestions on how to clean this up and get the installation working where my icons work.  Even if I do not get a working desktop icon, I would settle for a working Menu > Programs > Arduino IDE menu item.  

I am back to:
cd /home/pi/Downloads/
sudo ./uninstall.sh
rm /home/pi/Desktop/arduino-arduinoide.desktop
Deleted Menu > Programming > Arduino IDE
The directory '/usr/local/bin' is empty

Thanks!

dmcquestion, 2020-08-07:
Update:  The 32-bit version does not have these issues.

amahpour, 2021-07-20:
> Having these issues on version 1.8.13 on ARM64 running on a Raspberry Pi4 Model B w/ 4G RAM
> 
> Download and Extract arduino-1.8.13-linuxaarch64.tar.xz
> cd /home/pi/Downloads/arduino-1.8.13
> sudo ./install.sh
> Adding desktop shortcut, menu .....
> touch: cannot touch '/root/.config/mimeapps.list': No such file or directory /usr/bin/xdg-mime: 848: /usr/bin/xdg-mime: cannot create /root/.config/mimeapps.list.new: Directory nonexistent
> done!
> 
> I get the desktop icon 'arduino-arduinoide.desktop'
> 
> I execute the chown command:
> sudo chown pi /home/pi/Desktop/arduino-arduinoide.desktop
> 
> When I click on the desktop icon or go to Menu > Programming > Arduino IDE, I get the hour glass on my cursor for about 30 seconds and nothing.
> 
> This is my 9th or 10th installation attempt. I have tried multiple with the 64-bit. I have tried a couple with the 32-bit. I have tried using other instructions that have you move the arduino-1.8.13 to the /opt directory and try the installation from there. I have tried the installation with ./install.sh and sudo ./install.sh. All failed with various different errors and icon results.
> 
> I am open for suggestions on how to clean this up and get the installation working where my icons work. Even if I do not get a working desktop icon, I would settle for a working Menu > Programs > Arduino IDE menu item.
> 
> I am back to:
> cd /home/pi/Downloads/
> sudo ./uninstall.sh
> rm /home/pi/Desktop/arduino-arduinoide.desktop
> Deleted Menu > Programming > Arduino IDE
> The directory '/usr/local/bin' is empty
> 
> Thanks!

Try:
```
sudo chown UserName:UserGroup /home/username/Desktop/arduino-arduinoide.desktop
```

In my case my username is the same as the group so I did:
```
sudo chown $USER:$USER /home/$USER/Desktop/arduino-arduinoide.desktop
```

joeszem, 2021-07-27:
I had the same issue with the 32-bit and tried all these explanations. what ended up working was just CD to the correct file, then type: sudo ./install.sh

nayebare, 2022-11-04:
Hi @facchinm have you tried running this on  Ubuntu 20.04.5 LTS Ubunti 20.04.5 LTS. I run this command `sudo chown UserName:UserGroup /home/username/Desktop/arduino-arduinoide.desktop` but with no results. Any help is appreciated.

nhphuong91, 2022-11-04:
> Hi @facchinm have you tried running this on Ubuntu 20.04.5 LTS Ubunti 20.04.5 LTS. I run this command `sudo chown UserName:UserGroup /home/username/Desktop/arduino-arduinoide.desktop` but with no results. Any help is appreciated.

Did you replace <UserName>:<UserGroup> with your actual UserName & UserGroup?

nayebare, 2022-11-04:
> sudo chown yourusername /path/to/icon/arduino-arduinoide.desktop

Hi @nhphuong91 sorry this was the command i ran `sudo chown yourusername /path/to/icon/arduino-arduinoide.desktop`

nhphuong91, 2022-11-04:
> > sudo chown yourusername /path/to/icon/arduino-arduinoide.desktop
> 
> Hi @nhphuong91 sorry this was the command i ran `sudo chown yourusername /path/to/icon/arduino-arduinoide.desktop`

Then did you replace `<yourusername>` with your actual username?

nayebare, 2022-11-04:
Yes @nhphuong91 i did.

nayebare, 2022-11-04:
Hey @nhphuong91 this is the error i get when i try to run arduino in the terminal. `./arduino: line 35: /path/arduino-1.8.19-linuxaarch64/arduino-1.8.19/java/bin/java: cannot execute binary file: Exec format error`