arduino/Arduino #8785: [Library Manager] Please add IoTesla Client to Library Manager

CieNTi, 2019-04-13:


CieNTi, 2019-04-20:
Sorry to bump this issue, but will be possible to add this library to the manager?

For extra info: The library is initially blank because one of the part of the workshop I'll give is related to actually create the library, so the assistants will write the internal code for this library to work with the final system.

Tipically libraries proposals get any kind of answer within 4-5 days. There is some problem with this one? Please tell me so I can fix the required as soon as possible, as the workshop will start next week.

Big thanks

per1234, 2019-04-20:
Hi. There will be no problems with adding your library to the Library Manager index. It is fully compliant with the requirements.

As for the schedule of adding libraries, there is no fixed timeline. It happens as the developer in charge has time. I don't have the access to do the additions. Currently, it is a holiday weekend in Italy so I would expect Tuesday would be the earliest possible date, but with no guarantee.

CieNTi, 2019-04-21:
True about the holiday time hahaha

I just payed attention to "7 days ago" but didn't realize about the holy week/easter thing

Thanks for answering :D

cmaglie, 2019-04-23:
done, and sorry for the delay!

closed 2019-04-23 by cmaglie

CieNTi, 2019-04-23:
Big thanks!