arduino/Arduino #8792: Please add ELT CO2 Sensor S-300 series Arduino library

mhorimoto, 2019-04-15:
Hello.

I created a library to manage ELT CO2 Sensor S-300 I2C inteface for Arduino UNO.
Could you add this to a list of libraries?

URL:  https://github.com/mhorimoto/ELT_S300_HOLLY

Best regards;

mhorimoto, 2019-04-19:
I could not find this library in Arduino IDE 1.8.9 Library manager.
When I can find this library in Arduino IDE?

per1234, 2019-04-19:
Your library will not be available for installation via Library Manager until it has been added to the index. cmaglie occasionally comes through to clear all the outstanding Library Manager inclusion requests and closes those issues when that is done. You will receive a notification from GitHub of this and your library will be available in Library Manager shortly after that. So we'll get to it soon, please be patient.

cmaglie, 2019-04-23:
done

closed 2019-04-23 by cmaglie