arduino/Arduino #8795: Please add the IRMP library

ArminJo, 2019-04-15:
Hello,

I converted the IRMP sources to an Arduino compatible library.
https://github.com/ukw100/IRMP

Could you add it to your list of libraries?

Thank you.
Armin

TomasRoj, 2019-04-15:
Hey @per1234 how I can get permission to label these issues. I think I can help 🔖

TomasRoj, 2019-04-15:
@ArminJo to add your library to arduino package manager you must move your files from src folder to the root folder of you directory.

per1234, 2019-04-15:
>o add your library to arduino package manager you must move your files from src folder to the root folder of you directory.

@TomasRoj that is incorrect. Putting the source code files in the `src` subfolder, with a library.properties file in the library root folder is the 1.5 Arduino Library format. The IRMP library meets all the requirements for addition in the Arduino Library Manager index in its current form.

TomasRoj, 2019-04-15:
@per1234 oh sorry than :relaxed: I had same thing and you said that it's incorrect but thanks for reply. And hoe I can have this permission to add labels to issues?

per1234, 2019-04-15:
>I had same thing and you said that it's incorrect

No, your situation was different:
https://github.com/arduino/Arduino/issues/8710#issuecomment-477338770
your entire library was in a subfolder of the repository, not in the root of the repository as required. ArminJo's library is in the root of the repository; it's just that they have their source files stored in a subfolder of the repository, in accordance with the 1.5 Arduino library format. You could do the same with your library if you wanted (converting from 1.0 to 1.5 format) and it would still be compliant with the requirements of the Arduino Library Manager.

Please study the Arduino library specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification
I think it will make everything clear. If you still have some questions after that, we can discuss it on the forum.

>hoe I can have this permission to add labels to issues?

I think we have the labeling of issues under control already. Thanks for the offer though!

cmaglie, 2019-04-23:
done

closed 2019-04-23 by cmaglie