arduino/Arduino #8797: [Library Manager] Add SparkFun 9 DoF IMU - ICM 20948 - Arduino Library

oclyke, 2019-04-16:
Hello, 

Please add our ICM 20948 library to the library manager. Here is the link:
https://github.com/sparkfun/SparkFun_ICM-20948_ArduinoLibrary

Thanks!

per1234, 2019-04-17:
Hi. The library.properties is missing the required `paragraph` field. Please resolve this issue and then either create a new release (after updating the `version` field in library.properties) or else redo the existing release.

After that's done, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

oclyke, 2019-04-17:
@per1234 Thanks for catching that and letting me know! I added the paragraph field and incremented the release. 

All done

cmaglie, 2019-04-23:
done

closed 2019-04-23 by cmaglie