arduino/Arduino #8802: File not found error in windows ver 1.8.9, 1.8.8 works fine.

jwschull, 2019-04-18:
Here is the error:
Arduino: 1.8.9 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, Disabled, ck, 26 MHz, 40MHz, DOUT (compatible), 512K (no SPIFFS), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

java.io.IOException: The system cannot find the file specified
	at java.io.WinNTFileSystem.createFileExclusively(Native Method)
	at java.io.File.createTempFile(File.java:2024)
	at processing.app.BaseNoGui.saveFile(BaseNoGui.java:885)
	at processing.app.SketchFile.save(SketchFile.java:287)
	at processing.app.Sketch.save(Sketch.java:138)
	at processing.app.SketchController.save(SketchController.java:352)
	at processing.app.Editor.handleSave2(Editor.java:1923)
	at processing.app.Editor.access$1000(Editor.java:115)
	at processing.app.Editor$11.run(Editor.java:1903)
	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
The system cannot find the file specified

Rolling back to 1.8.8 and all is well.  NOTE:  Updating boards fails in 1.8.9 also.

facchinm, 2019-04-18:
Hi @jwschull ,
I just tested IDE 1.8.9 with `fqbn=esp8266:esp8266:generic:xtal=80,vt=flash,exception=disabled,ResetMethod=ck,CrystalFreq=26,FlashFreq=40,FlashMode=dout,eesz=512K,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none` and it compiled fine.
I can only think about an antivirus that was aware of IDE 1.8.8 and allowed it to write to `C:\Users\Username\AppData\Local\Temp\` while it is not allowing the newer 1.8.9, since `BaseNoGui.saveFile` routine didn't change between these two versions.

Would you mind disabling the antivirus for the time to execute a clean test? Thanks!

per1234, 2019-09-16:
@jwschull we're still waiting for you to let us know if disabling the antivirus resolved this issue for you (or if there was some other resolution you found). Please take a minute to provide us with an update.

per1234, 2020-09-25:
Closing due to lack of response.

@jwschull if you'll provide the requested information I'll be happy to re-open this issue.

closed 2020-09-25 by per1234