arduino/Arduino #8803: Building Arduino -

rtek1000, 2019-04-19:
Hi,

I've tried following the instructions in this link below, but it should be for compilation experts, because it is summarized.
https://github.com/arduino/Arduino/wiki/Building-Arduino

I see a part that says 1GB of download, but it only has 84MB. (So how to get all 1GB?)

> (Note: The download size is greater than 1GB and might hog all your bandwidth for a long while if your download speed is slow, such as a dialup connection or anything below 8mbps)

I suggest that the instructions be reviewed and, if possible, more detailed.

I found these instructions that may be useful for others who are not so familiar with this build type:

Windows:
https://www.instructables.com/id/Build-Arduino-IDE-From-GIT-Sources-on-Windows/

Ubuntu:
https://www.instructables.com/id/Build-Arduino-From-GIT-Sources-on-Ubuntu-1604/
 
Thank you.

per1234, 2019-04-19:
Thanks for pointing out that outdated information about the download size! The recommended `clone` command was updated to use the `--depth 1` option to make the download smaller. This is why your download was only 84 MB instead of > 1 GB, as it would have been without the `--depth 1` option.

I have updated the documentation accordingly:
https://github.com/arduino/Arduino/wiki/Building-Arduino/_compare/e418c070869ca1217619675ce45e2b24b0cf36b5...396466d9d3713086ebe76215367d195d33f15676

closed 2019-04-19 by per1234

rtek1000, 2019-04-19:
Hi,

Thank you.

I explained a little bit (the steps I needed to do):

```
(# are comments)
($ are commands, ignore the charactere $, because $ is the prompt from cygwin console, only for reference)
(e.g. meaning: for example)

# download java 8 JDK for Windows (e.g. "1.8.0_202", select correct version, x64 or x32) and install (in Windows system)
# to check: open CMD.exe and type: wmic os get osarchitecture
# https://www.oracle.com/technetwork/java/javase/downloads/jdk8-downloads-2133151.html
# to check: (after install) open the program from start menu (or run from WinKey+R) CMD.exe and type java -version

# download ant apache (select correct version for the java JDK version (8): e.g. apache-ant-1.10.5-bin.zip), and extract to Windows locale e.g. c:\apache-ant-1.10.5-bin
# https://ant.apache.org/bindownload.cgi

# download cygwin for Windows (select correct version)
# x64: https://www.cygwin.com/setup-x86_64.exe
# x32: https://www.cygwin.com/setup-x86.exe

# run cygwin setup-*.exe, (view: full; search by words, one by one), select the packages versions and install
# packages (select): git, make, mingw64-x86_64-gcc-g++, perl, unzip, zip
# default packages (check/select): coreutils, gzip, tar
# optional packages (select): openssh, nano

# run cygwin console (Desktop shortcut) and run commands below

# goto /home/user path (optional)
$cd ~

# default editor: vi (basic usage: https://www.cs.colostate.edu/helpdocs/vi.html) 
# to use nano editor (selected from cygwin setup):
$git config --global core.editor "nano"

# get sources
$git clone https://github.com/arduino/Arduino.git

# ($git clone... for 1st time only, or use: '$git pull' to get updates)

# set JAVA_HOME
# tip: run CMD.exe (from Windows) and type 'dir /x' to see short names
$export JAVA_HOME=/cygdrive/c/PROGRA~1/Java/jdk1.8.0_202/

# set ANT_HOME
# tip: run CMD.exe (from Windows) and type 'dir /x' to see short names
$export ANT_HOME=/cygdrive/c/APACHE~1.5-B/

# set PATH (multiple references need separator ':')
$export PATH=${PATH}:${ANT_HOME}/bin:${JAVA_HOME}/bin

# to check ant: after set PATH in cygwin type: ant -version

# go to /home/user/Arduino/build path
$cd ~/Arduino

# to compile (from /home/user/Arduino/build path)
$ant clean dist -Dplatform=windows

# to run
$ant run

# optional command for MAC: $ant clean dist -Dplatform=macosx

# optional command for Linux: $ant clean dist -Dplatform=macosx-java-latest
```

facchinm, 2019-04-19:
@rtek1000 as per today's `master`, the only needed tools are `ant` (to build) and `git` (to clone the repo). Cygwin is not needed anymore since we merged https://github.com/arduino/Arduino/pull/8682

rtek1000, 2019-04-19:
Hi,

Thanks for comment.

Is not Cygwin necessary or does it not work?

Since I was with the programs installed and the packages downloaded, I tried compiling (using Cygwin), but it can not run, it just opens the splash window quickly, and it does not open the IDE after the splash closes.

> BUILD SUCCESSFUL
> Total time: 14 minutes 52 seconds

[log-cygwin.txt](https://github.com/arduino/Arduino/files/3098818/log-cygwin.txt)

rtek1000, 2019-04-19:
Hi,

Without Cygwin returns error (PowerShell):

> PS C:\cygwin64\home\UserZ\Arduino\build> ant clean dist -Dplatform=windows
> Buildfile: C:\cygwin64\home\UserZ\Arduino\build\build.xml
> 
> subprojects-clean:
> 
> clean:
>    [delete] Deleting directory C:\cygwin64\home\UserZ\Arduino\arduino-core\bin
>    [delete] Deleting: C:\cygwin64\home\UserZ\Arduino\arduino-core\arduino-core.jar
> 
> clean:
>    [delete] Deleting directory C:\cygwin64\home\UserZ\Arduino\app\bin
>    [delete] Deleting: C:\cygwin64\home\UserZ\Arduino\app\pde.jar
> 
> linux-clean:
> 
> windows-clean:
>    [delete] Deleting directory C:\cygwin64\home\UserZ\Arduino\build\windows\work
> 
> macosx-clean:
> 
> BUILD FAILED
> C:\cygwin64\home\UserZ\Arduino\build\build.xml:381: java.nio.file.InvalidPathException: Illegal char <*> at index 52: C:\cygwin64\home\UserZ\Arduino\build\macosx\arduino-*.dmg
>         at sun.nio.fs.WindowsPathParser.normalize(WindowsPathParser.java:182)
>         at sun.nio.fs.WindowsPathParser.parse(WindowsPathParser.java:153)
>         at sun.nio.fs.WindowsPathParser.parse(WindowsPathParser.java:77)
>         at sun.nio.fs.WindowsPath.parse(WindowsPath.java:94)
>         at sun.nio.fs.WindowsFileSystem.getPath(WindowsFileSystem.java:255)
>         at java.io.File.toPath(File.java:2234)
>         at org.apache.tools.ant.taskdefs.Delete.isDanglingSymlink(Delete.java:879)
>         at org.apache.tools.ant.taskdefs.Delete.execute(Delete.java:615)
>         at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
>         at sun.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
>         at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
>         at java.lang.reflect.Method.invoke(Method.java:498)
>         at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:99)
>         at org.apache.tools.ant.Task.perform(Task.java:350)
>         at org.apache.tools.ant.Target.execute(Target.java:449)
>         at org.apache.tools.ant.Target.performTasks(Target.java:470)
>         at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1388)
>         at org.apache.tools.ant.Project.executeTarget(Project.java:1361)
>         at org.apache.tools.ant.helper.DefaultExecutor.executeTargets(DefaultExecutor.java:41)
>         at org.apache.tools.ant.Project.executeTargets(Project.java:1251)
>         at org.apache.tools.ant.Main.runBuild(Main.java:834)
>         at org.apache.tools.ant.Main.startAnt(Main.java:223)
>         at org.apache.tools.ant.launch.Launcher.run(Launcher.java:284)
>         at org.apache.tools.ant.launch.Launcher.main(Launcher.java:101)
> 
> Total time: 9 seconds
> PS C:\cygwin64\home\UserZ\Arduino\build>

cmd.exe:

> C:\cygwin64\home\UserZ\Arduino\build\build.xml:381: java.nio.file.InvalidPathException: Illegal char <*> at index 52: C:\cygwin64\home\UserZ\Arduino\build\macosx\arduino-*.dmg
>         at sun.nio.fs.WindowsPathParser.normalize(WindowsPathParser.java:182)
>         at sun.nio.fs.WindowsPathParser.parse(WindowsPathParser.java:153)
>         at sun.nio.fs.WindowsPathParser.parse(WindowsPathParser.java:77)
>         at sun.nio.fs.WindowsPath.parse(WindowsPath.java:94)
>         at sun.nio.fs.WindowsFileSystem.getPath(WindowsFileSystem.java:255)
>         at java.io.File.toPath(File.java:2234)
>         at org.apache.tools.ant.taskdefs.Delete.isDanglingSymlink(Delete.java:879)
>         at org.apache.tools.ant.taskdefs.Delete.execute(Delete.java:615)
>         at org.apache.tools.ant.UnknownElement.execute(UnknownElement.java:292)
>         at sun.reflect.GeneratedMethodAccessor4.invoke(Unknown Source)
>         at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
>         at java.lang.reflect.Method.invoke(Method.java:498)
>         at org.apache.tools.ant.dispatch.DispatchUtils.execute(DispatchUtils.java:99)
>         at org.apache.tools.ant.Task.perform(Task.java:350)
>         at org.apache.tools.ant.Target.execute(Target.java:449)
>         at org.apache.tools.ant.Target.performTasks(Target.java:470)
>         at org.apache.tools.ant.Project.executeSortedTargets(Project.java:1388)
>         at org.apache.tools.ant.Project.executeTarget(Project.java:1361)
>         at org.apache.tools.ant.helper.DefaultExecutor.executeTargets(DefaultExecutor.java:41)
>         at org.apache.tools.ant.Project.executeTargets(Project.java:1251)
>         at org.apache.tools.ant.Main.runBuild(Main.java:834)
>         at org.apache.tools.ant.Main.startAnt(Main.java:223)
>         at org.apache.tools.ant.launch.Launcher.run(Launcher.java:284)
>         at org.apache.tools.ant.launch.Launcher.main(Launcher.java:101)
> 
> Total time: 1 second
> 
> C:\cygwin64\home\UserZ\Arduino\build>

rtek1000, 2019-04-19:
Java and ant versions:

> C:\cygwin64\home\UserZ\Arduino\build>java -version
> java version "1.8.0_202"
> Java(TM) SE Runtime Environment (build 1.8.0_202-b08)
> Java HotSpot(TM) 64-Bit Server VM (build 25.202-b08, mixed mode)
> 
> C:\cygwin64\home\UserZ\Arduino\build>ant -version
> Apache Ant(TM) version 1.10.5 compiled on July 10 2018
> 
> C:\cygwin64\home\UserZ\Arduino\build>

rtek1000, 2019-04-22:
This splash error appeared to be in the preferences file. Just rename it to another name, and the IDE opened and created another file.

Thank you.