arduino/Arduino #8809: [Library Manager] Add ToneAC library

CornyjK, 2019-04-21:
I have emailed @teckel12 long time ago but he did not respond, I wanted to send a PR to his repository, but I am not allowed to by BitBucket. I had to create separate repository, where I met the requirements for adding to Arduino index.

Repo for index: https://github.com/RoboticsBrno/ToneAC
If teckel responds me, I am willing to send PR and add his repo to index.

teckel12, 2019-04-21:
@CornyjK I'm typically very prompt about these kinds of things. Maybe the email was blocked by spam filtering?

Anyway, maybe I should move my Arduino repos to Github for better access. I would also prefer to keep control of the library, as I use and still maintain it.

If I recall, someone else (but maybe you) did suggest changes for this purpose, but it seemed like it would break how I use the library. But maybe that was something different or someone else.

I'll move at least this repo to Github and then I'd love to see a pull request. I should have time later today.

CornyjK, 2019-04-21:
I'm glad to hear that. My changes would not break anything.

per1234, 2019-04-23:
@CornyjK @teckel12 how should we proceed with this request? Do you still want us to add https://github.com/RoboticsBrno/ToneAC to the Library Manager index, or should we wait until @teckel12 has moved their repository to GitHub and then add that one instead?

CornyjK, 2019-04-23:
I would wait for a while.

per1234, 2019-04-23:
OK, will do. Please comment here when ready to proceed with adding one of the repositories.

teckel12, 2019-04-24:
I'm just finding time to do it. At the end of marathon training so I don't have much free time.

teckel12, 2019-04-26:
@CornyjK Moved it to Github.  Still need to add the readme and the release, but it's up for you to submit a pull request.

teckel12, 2019-04-27:
Readme and release download are working.  Will be changing the creole readme to markdown, but it works for now.

per1234, 2019-04-27:
In order to be added to the Library Manager index, the library will need to have a library.properties file added to the root folder, and then a release needs to be made which contains that file.

Note that Arduino no longer accepts libraries to the Library Manager index if their library.properties `name` value starts with the word "arduino". It's no problem that the repository name starts with the word "arduino".

Once that's done, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

PaulStoffregen, 2019-04-27:
Oh, wow, this is the first I've heard of the new name policy.  Totally makes sense from a trademark perspective.  Is "arduino" allowed at the end of the name?  Was planning to submit a library soon with a "forArduino" suffix.

per1234, 2019-04-27:
The new policy only excludes the `name` value from starting with the word "Arduino" (I'm assuming case-insensitive). This is about reserving these names for use in official Arduino libraries. As Massimo Banzi said in the forum thread about this:
http://forum.arduino.cc/index.php?topic=604471
>Our official libraries from now on we'll be called Arduino_Something

Even though that would not strictly conflict with names like ArduinoSomething or Arduino-Something in 3rd party libraries, it might still result in confusion for the users.

I don't know what sort of trademark issues there might be from the use of the word "Arduino" in general. The only statement I'm aware of from Arduino on that subject is that [it should not be used for hardware](https://www.arduino.cc/en/Main/FAQ#toc9).

teckel12, 2019-04-27:
@per1234 I only start the repo with Arduino to keep things organized. The library itself has always just been named `toneAC`.

CornyjK, 2019-04-27:
'library.properties' will be included in my PR, I'll send it today.

teckel12, 2019-04-27:
@CornyjK I opened a new PR with your changes in the development branch instead of direct to master (just my typical git flow).  If you want to review the development branch and if you're happy with it I can merge it to master.

teckel12, 2019-04-30:
Merged to master, should be ready: https://github.com/teckel12/arduino-toneac

per1234, 2019-04-30:
The last thing needed for the library to be added to the Arduino Library Manager index is for the 1.5.0 GitHub release or Git tag to be created. Although the repository already has a 1.4 release, that release does not contain a library.properties file, and so it won't be picked up by the Library Manager indexer.

Once the tag/release has been created, please comment here.

teckel12, 2019-04-30:
@per1234 Done.

cmaglie, 2019-05-14:
done!

closed 2019-05-14 by cmaglie