arduino/Arduino #8812: [RAM Disk] build.path preference to another partition, but gives error

rtek1000, 2019-04-22:
Hi,

I'm trying to use a RAM disk to help with compiling, but it's giving error.

preferences.txt:
> build.path=Q:\arduino_build

In the forum they said about a folder to cache, but I could not make it work,

preferences.txt:
> build.cache=Q:\arduino_cache

Nor:
> build.cache.path=Q:\arduino_cache

It looks like the user folder is still being used (C:\Users\user).

'Q:\' is a 100MB RAM Disk (created by 'ROG RAMDisk 2.02.06').

Logs:
First time ok:
> Linking everything together...
> "I:\\Arduino\\1- IDE\\arduino-1.8.9\\hardware\\tools\\avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o "Q:\\arduino_build/barGraph.ino.elf" "Q:\\arduino_build\\sketch\\barGraph.ino.cpp.o" "Q:\\arduino_build/core\\core.a" "-LQ:\\arduino_build" -lm
> "I:\\Arduino\\1- IDE\\arduino-1.8.9\\hardware\\tools\\avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "Q:\\arduino_build/barGraph.ino.elf" "Q:\\arduino_build/barGraph.ino.eep"
> "I:\\Arduino\\1- IDE\\arduino-1.8.9\\hardware\\tools\\avr/bin/avr-objcopy" -O ihex -R .eeprom "Q:\\arduino_build/barGraph.ino.elf" "Q:\\arduino_build/barGraph.ino.hex"
> "I:\\Arduino\\1- IDE\\arduino-1.8.9\\hardware\\tools\\avr/bin/avr-size" -A "Q:\\arduino_build/barGraph.ino.elf"
> Sketch uses 1086 bytes (3%) of program storage space. Maximum is 32256 bytes.
> Global variables use 29 bytes (1%) of dynamic memory, leaving 2019 bytes for local variables. Maximum is 2048 bytes.

Second time error:
> Linking everything together...
> Rel: can't make C:\Users\user\AppData\Local\Temp\arduino_cache_475012\core\core_arduino_avr_uno_c94a07b2fc865919ed83bfeed74500ec.a relative to Q:\arduino_build
> Error compiling for board Arduino/Genuino Uno.

It seems to be related to this other issue of the link below (#8271), but apparently it's a different error: "Build folder in sketch folder fails on second verify", because the build folder is outside the sketch folder

facchinm, 2019-04-23:
Hi @rtek1000 ,
it looks like the issue comes from [this](https://github.com/arduino/arduino-builder/blob/29cf4c8b1473b4d0ba429c67ee497e2e061bd2fc/phases/linker.go#L58) line.
To bypass this in the short period you can disable "Aggressively cache compiled cores" (set `compiler.cache_core` to false in preferences.txt alternatively).

The cache folder path cannot be overridden right now via preferences, we should change [this](https://github.com/arduino/Arduino/blob/05de5c9593d6853e61a95ac0e91fc255267d49f3/arduino-core/src/processing/app/BaseNoGui.java#L266) line to allow this.

rtek1000, 2019-04-24:
Hi, ok, thank you for the attention!

cmaglie, 2020-02-12:
This should have been fixed some time ago by https://github.com/arduino/arduino-cli/pull/479, but I forget to close.

@rtek1000 please test and comment here if the issue is not solved for you.

closed 2020-02-12 by cmaglie

rtek1000, 2020-02-21:
Hi, congratulations, it looks like there has been a huge improvement in performance. When compiling a program without RAMdisk the time spent was approximately 18s and with RAMdisk it went down to about 14 seconds.

I noticed that some files stored on the HDD are still being lit during compilation. Would it be possible to copy these files to RAMdisk and redirect the IDE to no longer use the HDD?

**Warning: When trying to compile for STM32, an Access Denied error occurred on a RAMdisk folder. I am updating board support now, if the error persists I will inform you again.**

Thank you.

rtek1000, 2020-02-21:
> C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\arduino-builder -dump-prefs -logger=machine -hardware C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\hardware -hardware C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages -hardware C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\sketchbook\hardware -tools C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\tools-builder -tools C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\hardware\tools\avr -tools C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages -built-in-libraries C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\libraries -libraries C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\sketchbook\libraries -fqbn=STM32:stm32:GenF4:pnum=BLACK_F407VG,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,rtlib=nano -vid-pid=0403_6001 -ide-version=10812 -build-path Q:\arduino_buildbuild.verbose=true -warnings=all -build-cache C:\Users\robso\AppData\Local\Temp\arduino_cache_71324 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.STM32Tools.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\STM32Tools\1.3.2 -prefs=runtime.tools.STM32Tools-1.3.2.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\STM32Tools\1.3.2 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -verbose C:\Arduino IDE\arduino-1.8.10-windows-portable\examples\01.Basics\Blink\Blink.ino
> C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\arduino-builder -compile -logger=machine -hardware C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\hardware -hardware C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages -hardware C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\sketchbook\hardware -tools C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\tools-builder -tools C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\hardware\tools\avr -tools C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages -built-in-libraries C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\libraries -libraries C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\sketchbook\libraries -fqbn=STM32:stm32:GenF4:pnum=BLACK_F407VG,upload_method=swdMethod,xserial=generic,usb=none,xusb=FS,opt=osstd,rtlib=nano -vid-pid=0403_6001 -ide-version=10812 -build-path Q:\arduino_buildbuild.verbose=true -warnings=all -build-cache C:\Users\robso\AppData\Local\Temp\arduino_cache_71324 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.CMSIS-5.5.1.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\CMSIS\5.5.1 -prefs=runtime.tools.STM32Tools.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\STM32Tools\1.3.2 -prefs=runtime.tools.STM32Tools-1.3.2.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\STM32Tools\1.3.2 -prefs=runtime.tools.xpack-arm-none-eabi-gcc.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -prefs=runtime.tools.xpack-arm-none-eabi-gcc-9.2.1-1.1.path=C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1 -verbose C:\Arduino IDE\arduino-1.8.10-windows-portable\examples\01.Basics\Blink\Blink.ino
> Using board 'GenF4' from platform in folder: C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\hardware\stm32\1.8.0
> Using core 'arduino' from platform in folder: C:\Arduino IDE\arduino-1.8.12-windows\arduino-1.8.12\portable\packages\STM32\hardware\stm32\1.8.0
> cmd /c if not exist "Q:\\arduino_buildbuild.verbose=true\\sketch" mkdir "Q:\\arduino_buildbuild.verbose=true\\sketch"
> 'true\sketch' is not recognized as an internal or external command,
> 
> operable program or batch file.
> 
> exit status 1
> Error compiling for board Generic STM32F4 series.

rtek1000, 2020-02-21:
When compiling without RAMdisk there are no errors, the compilation log is attached:
[Log, NO RAMdisk, No errors.txt](https://github.com/arduino/Arduino/files/4238161/Log.NO.RAMdisk.No.errors.txt)

rtek1000, 2020-02-21:
@cmaglie , it looks like something still hasn't worked.

Thank you.

rtek1000, 2020-02-21:
I tried to compile for ESP32, ESP8266 and another core for STM32, there were no errors.

It seems that there is some incompatibility with the STM32 (Official) core and the use of RAMdisk.

[Log, RAMdisk, another STM32 core, No errors.txt](https://github.com/arduino/Arduino/files/4238223/Log.RAMdisk.another.STM32.core.No.errors.txt)

mrguen, 2020-06-16:
@rtek1000  did you manage to compile for Arduino core STM32 on RAMDISK?