arduino/Arduino #8815: [Arduino IDE: Boardmanager]: suppport activate or deactivate Board via config menue

hasenradball, 2019-04-24:
**Actual Situation:** 
Many developers work with different platforms (arduino, ESP8266, ESP32, ect.). Each platform supports many variants of boards which are listed in the `boards.txt` file.
If you work with arduino and ESP variants for example you will have more than 40 boards available in the board manager menue. This is very uncomfortable when you have to switch between the boards, because of the long scrolling list.

**Proposal**
Support a boards config menue which allows to check or uncheck the board you need/you are working with.
Unchecked boards will be commented in the `boards.txt` file for example.
If you have a new board you can comfortable add this by check it later on.
With this feature the Arduino IDE will become more comfortable for all users and the menues become more clear and well-arranged.

Many developers would appreciate such a feature!

per1234, 2019-04-24:
Closing as duplicate of https://github.com/arduino/Arduino/issues/6034

closed 2019-04-24 by per1234

hasenradball, 2019-04-24:
What about the idea to use the hide function in combination with a comfortable configmenue?

per1234, 2019-04-25:
By "hide function"? do you mean the hide property in boards.txt that I described [here](https://github.com/arduino/Arduino/issues/6034#issuecomment-432076577)?

hasenradball, 2019-04-25:
Yes! The hide function works fine.
But manually it is very uncomfortable to go through all boards.txt file.
Hide function with a small menue to check or uncheck which board should be hide or not.

hasenradball, 2019-04-25:
It is ok that hide= false is not possible, only hide=true is possible.

per1234, 2019-04-25:
>Hide function with a small menue to check or uncheck which board should be hide or not.

That proposal fits under the topic of https://github.com/arduino/Arduino/issues/6034. If we considered https://github.com/arduino/Arduino/issues/6034 to be solved by the `hide` property of boards.txt, we would have closed that issue as solved.

>hide= false is not possible, only hide=true is possible.

Correct. As I explained previously at https://github.com/arduino/Arduino/issues/6034#issuecomment-432076577, it's simply the presence of the property that matters. The value of the property has no effect.

hasenradball, 2019-04-25:
What about such a menue?
This menue function should work as:
- Read in all boards.txt files
- show the boards names with a check button
- if unchecked the hide function is added if not available or uncomment
- if checked the hide function is set.

per1234, 2019-04-25:
If you have any input on this topic, please submit it as comments on https://github.com/arduino/Arduino/issues/6034. This issue is only serving to fragment the information related to that topic so I'm going to lock it now.