arduino/Arduino #8816: Error code: ‘avrude: error reading signature data, rc=-67’, when updating code to Arduino UNO wifi rev.2

greenhouseALTEN, 2019-04-24:
Hi i would be really happy if someone could help me out here.

I have been working with some code on different Arduinos for a project at my work. Problem is that two of my Arduinos (Arduino UNO wifi rev.2) has stopped working due to an error code that is present whenever I try to upload code to it.
•	I have tried different code.
•	Used different computers to upload to it.
•	Both tried with Base Shield connected and without.
•	Tried with sensors and cables connected and without anything connected to the board (except the USB-cable).
•	I am have only been using real Arduino boards.
•	I have been cooperating with some colleagues to sort out the error and also google the error code but it did not give me any clues on how to solve this issue but there was also some others that have experienced the same error code.

Error code: ‘avrude: error reading signature data, rc=-67’

I still get the same error code. On the first Arduino I got this error code appeared after I have been using the device for about two months. When that board stopped working I replaced the board with another device (also an Arduino UNO wifi rev.2). I ran the same code on it for 5 days and then I got the same error code. So now two of my Arduino UNO wifi rev.2 boards are not working, since I am not able to upload any code to it.
- What are the odds for that?

Arduino board 1, used for 2-3 months approx. maximum 200-250 uploads of code cycles.
Arduino board 2, used for 5 days approx. maximum 60 uploads of code cycles.

I have also experienced that the Arduino program has been running slow sometimes, then when uploading the code again and/or plug/unplug the USB connector it was running at normal speed again, even though I did not do any changes to the program. 
The code I have mostly been running on my Arduino consumes about 51 % of its memory (25316 bytes). See screenshot below.
The problem seems somehow to be memory-related and in the code there are a lot of global variables, but still it should not destroy the board right?

**Error code:**
avrdude: bad response to enter progmode command: 0xa0
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "ATmega4809", rc=-67
avrdude: error reading signature data, rc=-67
avrdude: error reading signature data, rc=-67

**Sketch size**
Sketch uses 25316 bytes (51%) of program storage space. Maximum is 49152 bytes.
Global variables use 892 bytes (14%) of dynamic memory, leaving 5252 bytes for local variables. Maximum is 6144 bytes.

//Br Oliver

TimEllis2019, 2019-04-28:
I've been using my Wifi Rev 2 for about 6 months now and have the same issue.  I cannot talk to the board and get the same error messages.
-tim

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "Arduino Uno WiFi Rev2, None (ATMEGA4809)"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\telli\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\telli\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\telli\Documents\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=off -vid-pid=0X03EB_0X2145 -ide-version=10809 -build-path C:\Users\telli\AppData\Local\Temp\arduino_build_749796 -warnings=none -build-cache C:\Users\telli\AppData\Local\Temp\arduino_cache_516358 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.avrdude.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino14 -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino14 -verbose C:\Users\telli\Documents\Arduino\libraries\WiFiNINA\examples\Tools\FirmwareUpdater\FirmwareUpdater.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\telli\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\telli\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\telli\Documents\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=off -vid-pid=0X03EB_0X2145 -ide-version=10809 -build-path C:\Users\telli\AppData\Local\Temp\arduino_build_749796 -warnings=none -build-cache C:\Users\telli\AppData\Local\Temp\arduino_cache_516358 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avr-gcc\5.4.0-atmel3.6.1-arduino2 -prefs=runtime.tools.arduinoOTA.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\arduinoOTA\1.2.1 -prefs=runtime.tools.avrdude.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino14 -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Users\telli\Documents\ArduinoData\packages\arduino\tools\avrdude\6.3.0-arduino14 -verbose C:\Users\telli\Documents\Arduino\libraries\WiFiNINA\examples\Tools\FirmwareUpdater\FirmwareUpdater.ino
Using board 'uno2018' from platform in folder: C:\Users\telli\Documents\ArduinoData\packages\arduino\hardware\megaavr\1.6.26
Using core 'arduino' from platform in folder: C:\Users\telli\Documents\ArduinoData\packages\arduino\hardware\megaavr\1.6.26
Detecting libraries used...
"C:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DMILLIS_USE_TIMERB3 "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\cores\\arduino/api/deprecated" "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\cores\\arduino" "-IC:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\hardware\\megaavr\\1.6.26\\variants\\uno2018" "C:\\Users\\telli\\AppData\\Local\\Temp\\arduino_build_749796\\sketch\\FirmwareUpdater.ino.cpp" -o nul
Using cached library dependencies for file: C:\Users\telli\AppData\Local\Temp\arduino_build_749796\sketch\ESP32BootROM.cpp
Generating function prototypes...
"C:\\Users\\telli\\Documents\\ArduinoData\\packages\\arduino\\tools\\avr-gcc\\5.4.0-atmel3.6.1-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atme [... cut ...]
         PDI clock Xmega : 0 kHz

avrdude: Partial Family_ID returned: "mega"
avrdude: AVR device initialized and ready to accept instructions

Reading | avrdude: bad response to enter progmode command: 0xa0
avr_read(): error reading address 0x0000
    read operation failed for memory "signature"
avrdude: error reading signature data for part "ATmega4809", rc=-67
avrdude: error reading signature data, rc=-67
avrdude: error reading signature data for part "ATmega4809", rc=-67

avrdude done.  Thank you.

facchinm, 2019-04-29:
Hi @greenhouseALTEN and @TimEllis2019 , we are investigating the issue; would you mind moving it to https://github.com/arduino/ArduinoCore-megaavr/issues ? Thanks!

TimEllis2019, 2019-04-29:
Thanks for the reply Martino.  I opened a new issue at the link you sent.

-tim

closed 2019-04-29 by per1234

comiati, 2019-08-02:
Problem solved after I changed the CPU 
I think something is wrong with the 32u4 some bugs...... after many many writers something is happening, I found this error in  more than one PCB,