arduino/Arduino #8823: [Library Manager] Please Add Mechatro Library

teamong, 2019-04-27:
Motor Driver Library for L298x.
easy to drive and measure current each motor.
please add to library mamager.
thank you.
https://github.com/teamong/Mechatro

per1234, 2019-04-27:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

teamong, 2019-04-27:
Yes. I have created a first release.
thank you.

cmaglie, 2019-04-29:
done!

closed 2019-04-29 by cmaglie