arduino/Arduino #8824: At debug time .ino file used for compiling is not present

nbenm, 2019-04-27:
When I try to debug a .ino file compiled from Arduino ide, I can't list it's content. File is not present. I try to di this on an stm32 board, using a bluepill converted to black magic probe.
But the problem is not here.

Here's what I do:
I "cd" to the .elf directory. This is taken from Arduino console.

`arm-none-eabi-gdb -ex "target extended-remote /dev/cu.usbmodem80B0AFA61" Blink.ino.elf
`
```
monitor swdp_scan
attach 1
list loop
```
I have an error message
`9 
/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/arduino_modified_sketch_672400/Blink.ino: No such file or directory.`

/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/arduino_modified_sketch_672400 directory doesn't exist.
So it's impossible to list source code. I have an other code than looks like mine, but it's not the same. There are many differences.

So my question is how can do to make this file not removed, or is there a workaround?

facchinm, 2019-04-29:
Hi @nbenm ,
to properly debug an elf with gdb (and get the correct source level references) you need to save the sketch (either by ticking "save when verifying or uploading" in Preferences panel or by saving explicitly before compiling).
If you don't do so, the IDE will create a temporary file in `arduino_modified_sketch_` folder and remove it as soon as possible.

closed 2019-04-29 by facchinm

nbenm, 2019-04-29:
>  you need to save the sketch (either by ticking "save when verifying or uploading" in Preferences panel

This is already the case.

> or by saving explicitly before compiling

I have also tried this now

But it's the same. When I list source with source command, the file listed is not the original .ino file.

Here's the output:

```
/private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/hardware -hardware /Users/nb/Library/Arduino15/packages -tools /private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/nb/Library/Arduino15/packages -built-in-libraries /private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/libraries -libraries /Users/nb/Dropbox/arduino/libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10804 -build-path /var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/arduino_build_319349 -warnings=none -build-cache /var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/arduino_cache_397819 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=/Users/nb/Library/Arduino15/packages/stm32duino/tools/stm32tools/2019.3.31 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/nb/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /Users/nb/Downloads/Blink/Blink.ino
/private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/hardware -hardware /Users/nb/Library/Arduino15/packages -tools /private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/nb/Library/Arduino15/packages -built-in-libraries /private/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/AppTranslocation/3D3ECE21-ABC7-4CDE-B039-2AF603ED1672/d/Arduino.app/Contents/Java/libraries -libraries /Users/nb/Dropbox/arduino/libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10804 -build-path /var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/arduino_build_319349 -warnings=none -build-cache /var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/arduino_cache_397819 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=/Users/nb/Library/Arduino15/packages/stm32duino/tools/stm32tools/2019.3.31 -prefs=runtime.tools.arm-none-eabi-gcc.path=/Users/nb/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 -verbose /Users/nb/Downloads/Blink/Blink.ino
Using board 'genericSTM32F103C' from platform in folder: /Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31
Using core 'maple' from platform in folder: /Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31
Detecting libraries used...
"/Users/nb/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10804 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DARDUINO_ARCH_STM32 "-I/Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31/system/libmaple" "-I/Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31/system/libmaple/include" "-I/Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31/system/libmaple/stm32f1/include" "-I/Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31/system/libmaple/usb/stm32f1" "-I/Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31/system/libmaple/usb/usb_lib" "-I/Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31/cores/maple" "-I/Users/nb/Library/Arduino15/packages/stm32duino/hardware/STM32F1/2019.3.31/variants/generic_stm32f103c" "/var/folders/7w/_q0msgvn3l19tt52wt24rgqw0000gn/T/arduino_build_319349/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
[... cut ...]
 Arduino 1.6.0rc1
   Sketch uses 11,900 bytes (11%) of program storage space. Maximum is 108,000 bytes.
   Global variables use 2,592 bytes of dynamic memory.
 Ported to Maple from the Arduino example 27 May 2011 By Marti Bolivar
*/

void setup() {
    // Set up the built-in LED pin as an output:
    pinMode(PC13, OUTPUT);
}

void loop() {
    digitalWrite(PC13,!digitalRead(PC13));// Turn the LED from off to on, or on to off
    delay(500);          // Wait for 1 second (1000 milliseconds)
}
```

nbenm, 2019-04-29:
I don't understand.
Trying again and again, it works now!
I can see the original source.
Thanks @facchinm