arduino/Arduino #8826: Stepper Library Documentation Addition

doorgunnerjgs, 2019-04-28:
After spending a day and half trying to use a stepper motor with the Stepper library, I finally found the solution on a model railroading forum. In the discussion on "Stepper(steps, pin1, pin2, pin3, pin4) " there should be a note that the mapping of pins to stepper inputs is pin1 = N1, pin2 = N3, pin3 = N2, and pin4 = N4. It is DEFINITELY not obvious and not easy to find the answer.

animeshsrivastava24, 2019-06-09:
@doorgunnerjgs  @per1234  On https://www.arduino.cc/en/Tutorial/StepperOneStepAtATime , a reference is given to the following website which shows connection of stepper motor pins to  H-Bridge http://www.tigoe.com/pcomp/code/circuits/motors/stepper-motors/ but no direct reference is mentioned for connection of pins to Arduino. May I rewrite the code and post it for review?

per1234, 2019-06-10:
Hi @animeshsrivastava24 I think this is already documented under the "Bipolar Stepper Circuit and schematic" section of that page. You need to click the "Show images for the bipolar circuit and schematic" link to show the Fritzing and schematic. Am I misunderstanding your proposal?

animeshsrivastava24, 2019-06-11:
@per1234  Pardon for the misunderstanding in the previous discussion. Sure, I found that schematic, I diverted to the other reference page from the
> For more information about the differences of the two types, please take a look at Tom Igoe's page on stepper motors.

@per1234 Please guide me on how to fix this issue.

per1234, 2021-04-04:
I haven't been able to find any references about this N1, N3 ... thing so I think it will be impossible to take action on this request. Therefore, I'm going to close this.

@doorgunnerjgs if you will comment here with more information I'll be happy to reopen this.

closed 2021-04-04 by per1234