arduino/Arduino #8832: COMM ports not updated in Arduino

wurth70, 2019-04-30:
Arduino IDE not showing attached board comm port in "Tools>Port:" drop down menu. However, the port is recognized and displayed correctly in the lower right corner of the IDE screen. Please see attached file.
[image]

matthijskooijman, 2019-04-30:
AFAIK the bottom right only shows the most recently chosen port, which does not neccesarily mean that the IDE is seeing the port right now.

Regardless, you device manager clearly shows the port as present, so it should show up in the port list. IIRC that list is refreshed every time you open up the tools menu (or even the ports submenu), so I'm not sure what's going on here or how to debug this further.

PaulStoffregen, 2019-04-30:
Indeed this long-established behavior is confusing.  Now that we have DiscoverManager running as threads, perhaps the time has come to make the status bar update, so it doesn't give the impression disconnected hardware is connected?

Do proposed changes to Arduino's GUI still need to go through some special approval process?

PaulStoffregen, 2019-04-30:
Also, no idea why COM12 is appearing in the Windows Device Managers, but not the Ports menu.

@wurth70 - If this condition is still present, can you please try 2 things?

1: Restart the Arduino IDE.  When you run Arduino again, does COM12 properly appear?  This could indicate the bug may be within Arduino's serial port discovery process, but Windows itself is probably working as it should.

2: Restart your computer, ideally with a full cold (power off) reboot.  Many PCs leave USB powered up even when the computer is "off".  If COM12 did not reappear when restarting just the Arduino software, but does reappear after a full reboot, it could indicate a deeper problem with Windows or how Arduino's serial discovery interfaces with Windows.

wurth70, 2019-04-30:
Ok. 
Idea #1 did not resolve the problem. Had done that several times yesterday before submitting ticket.
Idea #2 did not resolve the problem until after several IDE restarts and plugging and unplugging the board. No clear method for finally getting the port to show up in the IDE. I did notice that Windows didn't respond in real time to the plugging and unplugging of the board. In fact, often the comm port didn't show up in the device manager until I started up the IDE. However, even though the comm port became active in Windows, the IDE didn't recognize it as seen in my previous attached screenshot.
Idea #3 Cold boot may desktop every night. This problem is not evident today and the device manager is responding in real time when I connect and disconnect the board and the IDE recognizes and displays the comm port.
NOTE: The difference between today's event and yesterday's event is that I had been working with another board that uses the CH341 driver. I could never get that driver to load on my 64bit machine. This could be causing the IDE to not query properly or whatever.  Still does not explain why IDE and Device Mngr can't agree and present all system ports in IDE port menu?

wurth70, 2019-04-30:
Hi Paul,
Posted reply to your requests before viewing my email. Let me know if you have any more questions or need me to do something else.
Roy

PaulStoffregen, 2019-05-01:
Oh these sorts of Windows issues are painful

> I did notice that Windows didn't respond in real time to the plugging and unplugging of the board. In fact, often the comm port didn't show up in the device manager until I started up the IDE. [...]

Any chance you can try this on another PC?  Ideally one with a fresh install of Windows 10.

Maybe there are Windows experts who know far more than I do, but my general experience is when software (or malware/spyware or overly aggressive anti-virus) starts interfering with how Windows detects hardware, I've never found any way to fix other than completely erasing the hard drive and reinstalling Windows and all software.

Sorry, I just don't know any other way.  With Linux, problems can also be very hard, but usually the troubleshooting process leads to understanding.  With Windows, I almost never learn anything.  Just "try setuff until something works" is the process.  Even when that does work, the end result is little or no improved understanding of why things went so wrong or how to fix in the future.  Wish I could help more, but other than suggesting the obvious but extreme way to reinstall Windows and all software, and maybe check this while adding everything back to see if any particular thing causes the problem again.

wurth70, 2019-05-01:
I understand. My hope was ring a bell within the Arduino development community to see if some kind of polling effort by the IDE would allow for the port handshake to occur once the system shows it active.  Also, the lower right corner information is misleading and I think would be better served doing a "board info query" to see if a board is on the other end of the line before indicating that a comms port is active on the selected board.... just a thought.  I'm really impressed with the Arduino platform and its continuing evolution..
Again, thanks for the response - consider it closed.