arduino/Arduino #8833: sprintf for floats to AVR cpus

dsyleixa, 2019-04-30:
the issue about sprintf for floats to AVR cpus is still unresolved.
IIRC, there once a patch has been published to any a Arduino AVR system file, but it's vanished and still not implemented to 1.8.8 (and/or 1.8.9, CMIIW).
So 
a) where is the patch to choose sprintf evaluation in the boards menu?
b) when will it finally be integrated to the standard Arduino IDE releases?

per1234, 2019-04-30:
You already made an issue about this a few months ago: https://github.com/arduino/Arduino/issues/8574

As before, I'm closing as duplicate of https://github.com/arduino/Arduino/issues/5033

It's not helpful to make duplicate issues. It only results in more clutter in the issue tracker.

Related: https://github.com/arduino/Arduino/issues/1719

closed 2019-04-30 by per1234

dsyleixa, 2019-04-30:
yes, I couldn't find it any more, cause it's obscured in the "closed" section.
You'd better keep it open until it's integrated in the Arduino standard releases! ;)

per1234, 2019-04-30:
>I couldn't find it any more

GitHub provides a very powerful search functionality that's quite easy to use. Here you can see all the issues you've opened in this repository:
https://github.com/arduino/Arduino/issues?q=is%3Aissue+author%3Adsyleixa

Note that 4/7 of them are duplicates.

>You'd better keep it open until it's integrated in the Arduino standard releases!

We already have an open issue on this topic. Once issue per topic is all we need. Any more than that only makes the issue tracker hard to work with.

dsyleixa, 2019-04-30:
ah, interestingly I now found, the only open one is locked
arduino locked and limited conversation to collaborators on 15 Mar 2018
perhaps unlock it anew?

per1234, 2019-04-30:
The discussion on this topic devolved to the point where it became counterproductive. I would like to think we could leave issues unlocked to allow any valuable new and unique input to be added to the thread, but I'm not convinced that the situation has changed in the time since the issue was locked. It may simply be that everything useful that can be said on the subject has already been said.

dsyleixa, 2019-04-30:
> 
> 
>  I would like to think we could leave issues unlocked to allow any valuable new and unique input to be added to the thread

+1
I double that!