arduino/Arduino #8842: M0 + M4 floating point formatting by sprintf ("%f") is missing

dsyleixa, 2019-05-02:
other than to Arduino Due and ESP8266, for M0 & M4 the floating point formatting by sprintf (%f) is missing,
both for fp32 (float) and for fp64(double).

could that please be fixed by default for all ARM cores?

dsyleixa, 2019-05-08:
no labels yet "feature request" and "platform: ARM" ?

per1234, 2019-05-08:
@dsyleixa I didn't know whether to consider it a feature request or bug report. I've now added the `feature request` label. As for "platform: ARM", we don't have such a label. We do have an `Architecture: SAMD` label. I suspect that would be appropriate, bur wasn't sure what you meant by "M4", since Arduino doesn't use a microcontroller with ARM M4 core in any of their boards that I'm aware of.

In the case this is only about Arduino SAMD Boards, then the issue needs to be moved to that repository:
https://github.com/arduino/ArduinoCore-samd

dsyleixa, 2019-05-08:
I'm not sure which platform exactly - 
I have both a M0 Board and a M4 Board by Adafruit, and for both platforms sprintf fails for floats.
Still gladly looking forwrard to M4 SAMD51 boards by Arduino, too... 8)

per1234, 2019-05-09:
Moved to https://github.com/arduino/ArduinoCore-samd/issues/407

closed 2019-05-09 by per1234

dsyleixa, 2019-05-09:
a pity, again to close an unresolved issue... :-/

per1234, 2019-05-09:
You opened the issue in the appropriate repository, which makes it more likely to be resolved. We only want one open issue per topic, so I closed this one.

dsyleixa, 2019-05-09:
I actually didn't move it but I doubled that, to be solved simultaneously in either repo, wherever it is solved faster.

per1234, 2019-05-09:
There is no way to solve the issue in this repository. This repository only contains the Java source code of the Arduino IDE's GUI.