arduino/Arduino #8845: [Library Manager] Please add library Oatmeal-Protocol

noporpoise, 2019-05-03:
Dear Team,

Please add our library to the Library Manager:
https://github.com/shielddx/oatmeal-protocol/releases/tag/1.0

Thanks,
Isaac Turner

noporpoise, 2019-05-05:
New release with `library.properties` fix is here: https://github.com/shielddx/oatmeal-protocol/releases/tag/arduino-v1.1

Thanks,
Isaac

cmaglie, 2019-05-14:
I've added the library to the indexer but there is a problem:

```
JOB 2253 - 2019/05/14 01:30:53 Error syncing library: Error while zipping library: Symlink not allowed: /home/jenkins/jenkins/jobs/Libraries/workspace/gitclones/github.com/shielddx/oatmeal-protocol/examples/ArduinoIDEExample/oatmeal_protocol -> ../../src
```

we do not allow symlinks (they may cause problem on filesystem that do not support them).
To fix this:
1) remove all symlinks
2) commit & push
3) delete and redo the last git tag

The library should be picked up automatically within an hour.

noporpoise, 2019-05-14:
Hi @cmaglie - thanks for those steps. 

We're using symlinks because as I understand it Arduino needs our project to exist in a `src` directory in the root of the repo, but compiling our examples with Arduino-Makefile requires that the library source can be found directly in a directory named `oatmeal-protocol`. Other than duplicating files to be in multiple places, is there a nicer fix than using symlinks?

Thanks,
Isaac

bengtmartensson, 2019-05-14:
> is there a nicer fix than using symlinks?

I suspect [VPATH or vpath](https://www.gnu.org/software/make/manual/make.html#Directory-Search) will do the job. Have not looked at the details of your problem though.

per1234, 2019-05-15:
>as I understand it Arduino needs our project to exist in a src directory in the root of the repo

This is incorrect. Having the source files under the `src` subdirectory and a library.properties file in the root of the library folder is the Arduino library 1.5 format, which is one supported option. However, the Arduino library 1.0 format, which is having the source files under the root of the library folder, is also supported for backwards compatibility with the many libraries which still use the 1.0 format.

For inclusion in the Library Manager index, having a library.properties file in the root of the repository is required for both 1.5 and 1.0 format libraries.

The only time the 1.5 format is required is if you're using the [`precompiled` field in library.properties](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format), which your library does not do.

noporpoise, 2019-05-15:
I think we can resolve these issues if we rename our Arduino library name (from `OatmealProtocol` to `oatmeal-protocol`) in `library.properties` to match our repo name. I've put together a PR to do that (https://github.com/shielddx/oatmeal-protocol/pull/9). If that would resolve these issues we can merge that PR and create a new release.

noporpoise, 2019-05-17:
We've merged a PR that removed symlinks and renamed the Arduino library to match the repo name (`oatmeal-protocol`). I've deleted and re-created the release at https://github.com/shielddx/oatmeal-protocol/releases/tag/arduino-v1.1
Please let me know if you need us to do anything else.

cmaglie, 2019-05-25:
renamed the library, now it's published correctly.

closed 2019-05-25 by cmaglie