arduino/Arduino #8849: IDE 1.9.0-beta: Greyed out port while using Nano

alve89, 2019-05-04:
Hey guys,

I think there's a problem with the 1.9.0-beta version: While I can choose the port for my Nano in 1.8.10 properly, the port-field is greyed out in 1.9.0-beta, so I can't even choose another port.

facchinm, 2019-05-14:
Hi @alve89 ,
thanks for the report. 

Can you paste the full debug log of the IDE? If you are on windows, launch the IDE with `arduino_debug.exe` to get all the output.

A few background: Beta uses only pluggable discoverers (https://github.com/arduino/serial-discovery and https://github.com/arduino/mdns-discovery); while there should be no issue with them, they are standalone programs, so pedantic Antivirus could disallow running them.
Your feedback is very useful since we need to decide whether or not to release this feature for everyone in master branch :slightly_smiling_face:

alve89, 2019-05-14:
Of course I can do, but I'm on macOS - what should I do then?

facchinm, 2019-05-14:
:smile: 
To get the full log from macOS you can launch `Arduino.app/Contents/MacOS/Arduino` .
Otherwise, you can try launching directly the `serial-discovery` file you'll find in `Arduino.app/Contents/Java/hardware/` . 
Attaching also the exact OS version of macOS you are using would help :wink:

ShakeThat, 2019-10-12:
I have this problem as well, on macOS 10.13.6. When I run serial-discovery manually and tell it SYNC_START it seems to work fine and does find the wchusbserial port I'm looking for:

```
{
  "eventType": "add",
  "port": {
    "address": "/dev/cu.ARPT",
    "label": "/dev/cu.ARPT",
    "prefs": {},
    "identificationPrefs": {},
    "protocol": "serial",
    "protocolLabel": "Serial Port"
  }
}
{
  "eventType": "add",
  "port": {
    "address": "/dev/cu.Bluetooth-Incoming-Port",
    "label": "/dev/cu.Bluetooth-Incoming-Port",
    "prefs": {},
    "identificationPrefs": {},
    "protocol": "serial",
    "protocolLabel": "Serial Port"
  }
}
{
  "eventType": "add",
  "port": {
    "address": "/dev/cu.wchusbserial1480",
    "label": "/dev/cu.wchusbserial1480",
    "prefs": {
      "productId": "0x7523",
      "serialNumber": "",
      "vendorId": "0x1A86"
    },
    "identificationPrefs": {
      "pid": "0x7523",
      "vid": "0x1A86"
    },
    "protocol": "serial",
    "protocolLabel": "Serial Port (USB)"
  }
}
```

When I set discovery.debug=true and run the arduino binary manually I get:

```
Set log4j store directory /Users/shakethat/Library/Arduino15
found discovery: serial -> {runtime.ide.path}/tools/serial-discovery
with preferencess -> {
  pattern = {runtime.ide.path}/tools/serial-discovery
}

found discovery: mdns -> {runtime.ide.path}/tools/mdns-discovery
with preferencess -> {
  pattern = {runtime.ide.path}/tools/mdns-discovery
}

serial: Starting: /Applications/Arduino 1.9.0.app/Contents/Java/tools/serial-discovery
mdns: Starting: /Applications/Arduino 1.9.0.app/Contents/Java/tools/mdns-discovery
mdns: thread exit other exception
serial: thread exit other exception
java.lang.NullPointerException
	at cc.arduino.packages.discoverers.PluggableDiscovery.run(PluggableDiscovery.java:78)
	at java.lang.Thread.run(Thread.java:748)
java.lang.NullPointerException
	at cc.arduino.packages.discoverers.PluggableDiscovery.run(PluggableDiscovery.java:78)
	at java.lang.Thread.run(Thread.java:748)

```

If I manually set the port name in preferences.txt, uploading works properly (the Port menu remains grayed out).

ShakeThat, 2019-10-13:
OK, I believe the problem is that the space in
`/Applications/Arduino 1.9.0.app/Contents/Java/tools/serial-discovery` causes the exec to fail. I've renamed `Arduino 1.9.0.app` to `Arduino1.9.0.app` and port discovery is now working.

myelin, 2020-12-13:
Confirming the above comment: I had the same problem with the 1.9.0 beta downloaded today (2020-12-12), and removing the spaces from the path fixed it.

I unpacked the zip and moved Arduino.app to /Applications/Arduino 1.9.0 BETA.app, and the Port menu was greyed out.

Renaming it to /Applications/Arduino-1.9.0-BETA.app, now the Port menu works as expected.